svenhb/GRBL-Plotter #299: Hi, I try to find a lot settings but it is going to stuck when click any where on that window.

Makasudali, 2022-07-20:
I want to know can I take two words independently at one time with Gcode. So I am easily change size and location of that two words. and font also. Please go through img.
[image]

svenhb, 2022-07-20:
1. Create your first text and "Create GCode"
2. Enable "Add imported graphic to 2D view"
3. [image]
4. Create your second text and "Create GCode"

Makasudali, 2022-07-20:
> 1. Create your first text and "Create GCode"
> 2. Enable "Add imported graphic to 2D view"
> 3. [image]
> 4. Create your second text and "Create GCode"

Thanks for your help its working fine.