svenhb/GRBL-Plotter #302: Update to newer .net framework

rasyidf, 2022-07-21:
**Describe the solution you'd like**
Upgrading this project will benefits many fixes and performance it bring with newer frameworks. the're atleast 3.8k lines of improvement that can be applied with newer Analyzers.

[image]

**Describe alternatives you've considered**
for a starting point, we can update to .net 4.8 and separate the program (especially UI and Core) step by step. then we could, change the WinForm to .net core / .net 6 later.

**Additional context**
[Accesibility](https://docs.microsoft.com/en-us/dotnet/standard/analyzers/portability-analyzer)
[image]

svenhb, 2022-07-21:
Then GRBL-Plotter is not compatible with windows XP anymore.

svenhb, 2022-07-21:
Also I don't care most of the warnings. Some how you can configure the warnings, to get much less.

rasyidf, 2022-07-27:
and? will there be any update for net framework 4.8?

svenhb, 2022-07-27:
we'll see, maybe I'll start at the end of the year.

smakhtin, 2024-05-12:
I started the [PR here](https://github.com/svenhb/GRBL-Plotter/pull/398), to fix DPI autoscaling we need to update to .NET 4.8. Autoupdate seemd to work quite straightforward.

svenhb, 2024-05-12:
As already written: Then GRBL-Plotter is not compatible with windows XP anymore.
Will check how to work with new branch...
Thanks so far....

smakhtin, 2024-05-13:
@svenhb The PR title was about .net version update, so I linked PR here. In the app manifest, I marked support till Win7. Windows XP indeed is too old to support.