svenhb/GRBL-Plotter #305: What does PN in COM CNC menu mean?

dchicchon, 2022-07-25:
Currently I am debugging my plotter since it's not working and just wanted to check PN means in the COM CNC config menu? I haven't found the meaning of it in the documentation.

svenhb, 2022-07-25:
I assume you mean this area?: 
[image]

It is the Input Pin State: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#real-time-status-reports

-Pn:XYZPDHRS indicates which input pins Grbl has detected as 'triggered'.

-Pin state is evaluated every time a status report is generated. All input pin inversions are appropriately applied to determine 'triggered' states.

- Each letter of XYZPDHRS denotes a particular 'triggered' input pin. 
--  X Y Z XYZ limit pins, respectively
-- P the probe pin.
-- D H R S the door, hold, soft-reset, and cycle-start pins, respectively.
-- Example: Pn:PZ indicates the probe and z-limit pins are 'triggered'.
-- Note: A may be added in later versions for an A-axis limit pin.

dchicchon, 2022-07-26:
I see, so in my case since I see XYZ for Pn does that mean that the program can detect those pins are triggered? As in currently providing some sort of signal?

svenhb, 2022-07-26:
> I see XYZ for Pn does that mean that the program can detect those pins are triggered?

Yes the program recognizes these pins as triggered. Perhaps your connected hardware is doing this.
If I just connect a plain arduino nano, without any connected hardware, the signals are off (default config, $5=0, $6=0).

You could invert the signal via $5 and $6: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#grbl-settings

>is there a better place to discuss debugging my code

For me it is ok, perhaps I should open the Discussion option in the menu...

closed 2022-08-11 by svenhb