svenhb/GRBL-Plotter #306: A question about the magnification function

lianzaozi, 2022-08-01:
I'm using version 1.6.6.6, and when I zoom in, I notice that the cursor suddenly goes back to the coordinate (0,0) when I zoom in enough.

The DXF file I used in the test and my operation process are in the attachment
[test1.zip](https://github.com/svenhb/GRBL-Plotter/files/9234004/test1.zip)

svenhb, 2022-08-11:
Please check if problem is solved: https://github.com/svenhb/GRBL-Plotter/releases

lianzaozi, 2022-08-12:
> Please check if problem is solved: https://github.com/svenhb/GRBL-Plotter/releases

The source code of version 1.6.7.0 will report an error when compiling
[image]
[image]

Version 1.6.7.0 software zoom function problem has been solved, I will do more testing

svenhb, 2022-08-12:
Sorry, forgot to upload this files, before generating this relase (means, they are missing in the source.zip)
App. config, Settings.cs, Program.cs, GRBL-Plotter.csproj
in https://github.com/svenhb/GRBL-Plotter/tree/master/GRBL-Plotter

closed 2022-11-05 by svenhb