svenhb/GRBL-Plotter #308: Startup

JohnTurn, 2022-08-14:
Running Win10 Pro 64bit.  Installed GRBL-Plotter using setup program.  Running GRBL on a uno which is recognised by the Arduino IDE. When starting GRTBL-Plotter the window for selecting the COM port disappears after about a second before entering the main program.  No time to select com port.  This also happens when running the program from the extraced zip folder.
When running the program in a WIN10 sandbox on the same machine it behaves itself but the Windows sandbox is unable to provide the COM port so does not show up..  Installed on a second machine and it does the same thing.  Disabledd fire walls and virus checking to see if they are blocking the correct action with no change. 
Has this issue come up before?  Any othe rdetails needed?
Any suggestions gratefully recieved.

John

svenhb, 2022-08-14:
I assume the connection was already established an then the COM window minimized, did you check the taskbar?
[image]

JohnTurn, 2022-08-14:
Thank you.  Have just found what you have sugested myself. The program was finding COM1 and then disappearing only to be found  as you have said by looking at the taskbar and reopening the window.  Closing the COM1 conection anfd then selecting the USB Com port associated with the GRBL controller on the Nano.
All goes.

Thank you

closed 2022-08-14 by JohnTurn

svenhb, 2022-08-14:
👍👍👍