svenhb/GRBL-Plotter #309: Italian language

78Matteo78, 2022-09-20:
Hello How can I help you to have the Italian language too? greetings matteo

svenhb, 2022-09-21:
I like to use https://github.com/HakanL/resxtranslator/ , when adding a new language:
After creating the new *.resx files, they must be added to the visual studio project.

78Matteo78, 2022-09-21:
I try, but I think it will be difficult for me

svenhb, 2022-09-21:
At least to translate and create the file is not difficult, but a lot of work.
Copy the GRBL-Plotter project to a location and open via Recource Translator.
Add a new language
Insert translation of the "NoLanguageValue"
In most cases it is possible to select and copy more than one cell, translate it via google and copy back to new language cells. It doesn't work if there is a line-break within one cell, which will cause a new cell-row on paste.
[image]

[image]

[image]

78Matteo78, 2022-09-21:
does it only work with visual studio? I'll try to download it ...

svenhb, 2022-09-22:
Just to create the new resx files doesn't need visual studio, just the resx translator.
Send me the new files and I will make a new GRBL-Plotter version.

78Matteo78, 2022-09-23:
ok i managed to understand how it works. it will take me some time, once I have created the new files, I will send them to you by mail. Thanks for your help. Greetings Matteo

78Matteo78, 2022-09-25:
Hello i translated all resx files except one, main form in GUI because it gives me the following error in the picture
[image]

78Matteo78, 2022-09-25:
The image in to mail

svenhb, 2022-09-25:
Got no mail up to now

78Matteo78, 2022-09-25:
I'm sorry I was wrong ... now it should have arrived

svenhb, 2022-09-30:
Thanks for your work: https://github.com/svenhb/GRBL-Plotter/releases/tag/v.1.6.7.2

closed 2022-09-30 by svenhb