svenhb/GRBL-Plotter #310: "Z-Height engrave must be negative" causing problem when z-axis directed down

kirevdokimov, 2022-09-25:
I have a plotter where z-axis directed down, so 
- Pen down is like `Z 25`, 
- Pen up is like `Z -25`, 

but I cannot set the values to the setup because pen down should be negative and pen up should be positive. 

How to handle this situation?

[image]

kirevdokimov, 2022-09-25:
Oh, I see I can simply use Individual commands to set up what does pen up and pen down means
[image]

closed 2022-09-25 by kirevdokimov

svenhb, 2022-09-25:
What's about inverting the motion direction via $3? To have the "normal" CNC behavior?
https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#---view-grbl-settings

kirevdokimov, 2022-09-26:
Oh wow, `$3=4 (dir port invert mask:00000100)`   worked for me, thank you @svenhb :)