svenhb/GRBL-Plotter #311: UI became a little messy on

kirevdokimov, 2022-09-26:
Initially it loads upscaled blurry as an old legacy app. Not sure if you can notice this with a screenshot itself
[image]

But when I load an svg the window reloads as HDPI, but some font sizes are messed a little
[image]

Service buttons are below the frame so unaccessable
[image]

Current monitor resolution is 3840x2160

kirevdokimov, 2022-09-26:
This scaling problem can be kind of fixed by user by overriding compatibility options, especially setting high DPI scaling behavior to "System". The app will be blurry and looks like a legacy, but at least not messy

[image]

closed 2022-09-26 by kirevdokimov

svenhb, 2022-09-26:
Yes, there is problem with this scaling and I don't know how to fix it, see also here: https://github.com/svenhb/GRBL-Plotter/issues/239

Thanks for the solution.