svenhb/GRBL-Plotter #313: Tangential Knife and pen Cont.

jbaker1311, 2022-10-11:
I've done what you've recommended and now the pen and cutter are working
although in pen "mode" it works fine but when cutting both solenoids are activated.

[image]
[image]
[image]
[image]
[image]
[image]

svenhb, 2022-10-11:
I think you need to switch off both via "M9", before selecting the new seleonid:
try "G54;M9;M7" and "G55;M9;M8" for your commands.

jbaker1311, 2022-10-13:
I have done this, I think because my pen is pinned to spindle on/off this is why it's still operating while Mist is activated
I've changed to have pen on flood and cutter on Mist. It's till doing the same thing. Will go over our old thread and reply

svenhb, 2022-10-13:
I assume you use a servo for the pen-up-down and another servo for the cutter-up-down.
Perhaps you need to switch the pwm-signal to one OR the other servo, using a relais (triggered by the mist-signal)?

jbaker1311, 2022-10-13:
No I'm using compressed air, with solenoids that run off Mist and Flood signals.
Is there a simple way to keep laser on but independently turn Mist and Flood on and off

svenhb, 2022-10-13:
>Is there a simple way to keep laser on but independently turn Mist and Flood on and off
Is it possible if you try it manually? Is then working?
http://linuxcnc.org/docs/html/gcode/m-code.html#mcode:m7-m8-m9
M9 - mist and flood off 
M9;M7 only mist
M9;M8 only flood
M7;M8 mist and flood

jbaker1311, 2022-10-13:
I'll be back at the machine tomorrow, so will give that a crack.
Do I put it in the tool table?

jbaker1311, 2022-10-13:
I have them working separately now, the next problem I'm encountering is they are no longer going up and down.

jbaker1311, 2022-10-13:
[image]
[image]
[image]

jbaker1311, 2022-10-14:
So I've figured out that M3 and M5 are for spindle direction etc.
Not relevant in my situation, I can use M9 to deactivate my solenoids and raise pen. But when i have G55 M8; G54 M7 for pen down, it activates both. I think I need a way to have different commands per each tool.
Maybe an addition to software?
[image]
[image]

svenhb, 2022-10-14:
Perhaps you can describe your mechanics again? I don't get it...
You have a pen, needing "selection" and "up/down"
You have a cutter, needing "selection" and "up/down"
And also a laser?

You could use other hardware / grbl-version: https://github.com/fra589/grbl-Mega-5X/wiki
It has additionally:  Digital output control: M62, M63, M64, M65

jbaker1311, 2022-10-14:
My machine is a 2d Plotter, I've recommissioned an old industrial plotter made in the late 70's
It's basically an Arduino mega with the mega version of grbl
It has 2 tools that move on a gantry, a static pen and a cutting wheel/blade on a 360° z axis
The tools are air assisted for lifting and compression respectively. Before I gave up on the cutting wheel and just used the pen on the spindle enable and hand cut my job. 
I want use the table to full capacity so I'm trying to sort out the cutting side of it.
Now following our thread I've changed them to Mist/Flood
In the individual commands, I'm using M9 to lift the tool (pen up) then g54 M7: G55 M8 (pen down.) 
I'm asking if there is a command for pen down for a specific tool. E.g. pen down at start, draws, M9 command to lift, then moves to next pos, runs pen down which then runs both commands at once (G54 M7: G55 M8) which makes both tools lower

svenhb, 2022-10-14:
You need a logic connection of the general up/down signal (M3/M5) and the tool selection (M7/M8).
Tool 1 should be activated via the signals SPINDLE AND MIST
Tool 2 should be activated via the signals SPINDLE AND FLOOD

Perhaps it works with simple diodes:
[image]

But a real circuit would be better (74xx08)

jbaker1311, 2022-10-14:
I'll have a try at the 5x version of grbl and repin with the new digital toggles. At this stage I don't want to add new hardware

jbaker1311, 2022-10-16:
I've loaded the 5X version, now I'm not too sure what Command i should be sending to activate

jbaker1311, 2022-10-24:
Good day Sven,
I've made a breakthrough with my commands problem. Is there a specific way to indicate to the software that I only want a specific tool. 

As it is now, I've repinned to flood and coolant with m9 as a pen up command, and m7/M8 pen down. When paired with G55 M8;G54 M7 it will still activate both at the same time. Although when I go through gcode and delete the specific code that's unwanted it performs perfectly. 

My idea is to somehow tell the gcode creator the when G55 is the tool required to only read that specific pen down command rather than both G55 and G54

svenhb, 2022-10-24:
You need different pen-up/down commands, depending on selected tool.
Most simple: do a logic AND: (spindle and mist signal) and (spindle and not-mist signal) for the individual tool pen_up/down.
Most complicate: I need to implement individual commands, depending on selected tool-nr. Up to 255 tools possible with 2 states each...

svenhb, 2023-01-05:
I hope you could manage it with the external logic

closed 2023-01-05 by svenhb