svenhb/GRBL-Plotter #314: dxf line conversion

lfmmd, 2022-11-01:
I entered a dxf file dxf file, in the dxf file I used a dashed line when converting gcode he converted it to a solid [line]()

svenhb, 2022-11-01:
Sorry it only works for straight lines

svenhb, 2022-11-02:
I will try to fix that, perhaps you can upload your DXF file?

lfmmd, 2022-11-03:
@svenhb thank you for your support  I uploaded my dxf file and the reason I speculate is: in dxf the dashed line type is a
 Line description, and some display parameters have been added  
[1.zip](https://github.com/svenhb/GRBL-Plotter/files/9926139/1.zip)

svenhb, 2022-11-03:
The problem is up to now: I apply the dash pattern to one line segment. If the line segment is shorter than dash-pattern, no pattern is visible. Arcs and circles are converted to many short line segements.
My solution I#M working on: continue the dash pattern over many line-segments.

svenhb, 2022-11-05:
Try latest release: https://github.com/svenhb/GRBL-Plotter/releases

closed 2023-01-05 by svenhb