svenhb/GRBL-Plotter #315: Camera mount on XY or Z platform "Also scale on rotation not working"

gorgoi, 2022-11-17:
Hello svenhb, 

First of all, thanks for your work is a greate software with a lot of potential. 
I tried for several weeks to configure a XY camera with **automatic fiducial correction** and **also scale on rotation**.

I have to cut some plastic that doesn't have constant dimensions.

Problem
-------------------------
When I start **Automatic fiducial correction**, after first red circle recognition,  **enable shape recognition** for second one is deactivated, and then correction doesn't work. 
If I **enable shape recognition** manually rapidly before reaching second one, it see circle and fiducial but scaling/corection  also not working. **Video** attached, down the page.

It's only works for fixed mount camera(example file from you), but my plastic sheets are to large like 600mm to be recoined on fixed camera. 

File:
[image]

Video:
https://user-images.githubusercontent.com/24373830/202392908-b71a33c8-1127-4187-9cc6-72bc3a4eb6c9.mp4

Thanks for your support!

svenhb, 2022-11-17:
I'm a bit out of the topic and need some time to get used to it again

Did you see this "video" how it should work? https://grbl-plotter.de/data/uploads/pic/plotter_fiducials/fiducial_xy_auto.gif

In your video, there is already a fiducial assined, when starting, you may need to clear this list first (right-click in video-picture):
[image]

[image]

gorgoi, 2022-11-17:
Yes, is true. In video, I pressed **Start automatic fiducial correction**, at  second point,  for redo the procedure recognition. 

I will try again tonight and try to capture the screen for a clear view. 

Thanks for your fast replay. It is  a grate help.

gorgoi, 2022-11-18:
Hello again, 

I've checked again last night, and it seems that the SVG file that I have is not working maybe because it has bigger dimension?, i don't know..

Do you have any limit to **automatic fiducial correction**?... maybe this is the reason. 
Attached I  add it my test. 

**Dimensions:**
[image]

**SVG files_used in videos**
[Files.zip](https://github.com/svenhb/GRBL-Plotter/files/10039167/Files.zip)

**Video my file_NOK**

https://user-images.githubusercontent.com/24373830/202649040-a3f7f39b-bc49-4224-a231-55069f9ee182.mp4

**Video your default file OK**

https://user-images.githubusercontent.com/24373830/202648383-c68cd295-03ba-4b54-8ce8-703529bfeaaa.mp4

svenhb, 2022-11-18:
Thanks for the detailed information. 
I assume the time, reaching the 2nd fiducial, is too long and a timout happens.
Please check the logfile: look in in the  About box (menu-strip "About") for the data-path, open /logfiles/logfile.txt
If timout happened, an error should be logged: "Fiducial detection: 2, Grbl-Idle not reached" - is this correct?

svenhb, 2022-11-18:
...but the timeout can't be the reason for clearing the checkbox...
I'm still investigating...

gorgoi, 2022-11-18:
I was looking at the videos to make a difference, I see  coordinates for  **second fiducial**  picture red  **X:320.00, Y:240.00**   , that looks out of range/ limit..., i don't know what it is.., maybe is has something to do with the problem?... 

**File OK**  (the small one your SVG)seems to be in the range:
1.
[image]

2.
[image]

-----------------------------------------------

**File NOK**: seems to be out of range **(picture in RED)**
1.
[image]

2. 
doesn't go to next step.

------------------------------------------------
PS: I will check tonight "Please check the logfile: look in in the About box (menu-strip "About") for the data-path, open /logfiles/logfile.txt", when I will be at the computer from home. 

Thanks for your support!

svenhb, 2022-11-18:
This:
[image]
is the last shown message, before the timeout can occur - unfortunatly I forgot to show a message on this place.

I make a new version, correcting this bug, if you could confirm the error in the logfile

svenhb, 2022-11-18:
Perhaps you can increase moving speed a lot (just for test) to minimize traveling time and avoid timeout...

svenhb, 2022-11-18:
I mean G0 feedrate, set via $110 / $111

svenhb, 2022-11-18:
This checkbox is not needed for the automatic fiducial correction, and will be disabled after the automatic correction.
[image]

gorgoi, 2022-11-18:
Hello, 

Sorry, I'm with the speed at the limit, I can not go forward, it is screw driven. 
I found logs that has been  generated when videos were made, to have the same timeline. Please see:

File **NOK**, first video:
[log_20221117.4_my file_NOK.txt](https://github.com/svenhb/GRBL-Plotter/files/10043498/log_20221117.4_my.file_NOK.txt)

File **OK** scound video:
[log_20221117.5_your file_OK.txt](https://github.com/svenhb/GRBL-Plotter/files/10043501/log_20221117.5_your.file_OK.txt)

Thank you for your time!

svenhb, 2022-11-18:
I uploaded a new version here: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_6_7_4_test.exe
I hope bug is fixed.

gorgoi, 2022-12-01:
Thank you **svenhb**!,  I can confirm that is working. Since I used your test software, I had no problem.

closed 2023-01-05 by svenhb