svenhb/GRBL-Plotter #316: Unble to compile solution

LeeRossCritchlow, 2022-11-20:
Screen capture of Errors, Warning, Message.
[image]

LeeRossCritchlow, 2022-11-20:
I noticed that I still had vMicro installed. I uninstalled the extension, cleaned solution, rebuild solution. One Error & Warning remained.
[image]

LeeRossCritchlow, 2022-11-20:
I cleaned and rebuilt the solution twice. Each time the attached screen capture occurred.
[image]

LeeRossCritchlow, 2022-11-20:
ReSharper is reporting 35 errors in 5 files.
[image]

svenhb, 2022-11-20:
Where did you download the source?
I don't have a file "Bootstrapper.cs" and my projects name is GRBL-Plotter

svenhb, 2022-11-21:
Most probably you try to compile this work: https://github.com/IlyaChernov/grbl-Master/tree/master/grbl.Master.UI
I'm sorry I can't help

LeeRossCritchlow, 2022-11-21:
The attached Source shows the link on the page shown in Source1.
[image]
[image]

svenhb, 2022-11-21:
Strange, if I click on the link on this page https://github.com/svenhb/GRBL-Plotter , it opens the corresponding releases-page: https://github.com/svenhb/GRBL-Plotter/releases/tag/v1.6.7.3
And the dowloand of GRBL-Plotter-1.6.7.3.zip shows the correct content.

svenhb, 2022-11-21:
I think issue-removal is not possible - just closure.

closed 2022-11-21 by svenhb