svenhb/GRBL-Plotter #317: Using Gcode transform-radius compensation does not work correctly.

Atlan4, 2022-12-16:
I have a dxf file. after opening the GRBL plotter, I need to generate g code so that the dimensions change when using a 2mm milling cutter (or another size)

using Gcode transform-radius compensation does not work correctly.

As possible, achieve that the generated code has the dimensions of the holes reduced by 1 mm. well thank you

[image]
[image]

svenhb, 2022-12-17:
You may can upload the original DXF?
I assume the "bad" object consists of not connected lines.
You may try to connect the single lines via GRBL-Plotter by increasing this value in setup:
[image]

Atlan4, 2022-12-17:
Yes, your solution worked, even though the lines connected in the eagle field probably had different directions. after redrawing, it already displays normally even without changing the settings. (but it's good that it can be adjusted with the settings, I don't want to redraw it every time)

BUT, as you can see in the picture, the creature cuts from the inside and my special shape from the outside. Is there a solution to this? well thank you
[image]
[Panel cnc zadny v1.dxf.txt](https://github.com/svenhb/GRBL-Plotter/files/10251709/Panel.cnc.zadny.v1.dxf.txt)

svenhb, 2022-12-17:
>the creature cuts from the inside and my special shape from the outside

So the path direction of one is clockwise, from the other counter clockwise. 
If you enable the "Pen-up" path view (press space bar) you see the start direction.
When object is selected, you can reverse the direction - before applying the path offset:
[image]

Atlan4, 2022-12-17:
Thank you, it seems that it will work. I will definitely follow the rules when drawing, if I don't want to fix it later. Thank you again
[image]

svenhb, 2022-12-17:
Good to hear, that it works now.

closed 2023-05-01 by svenhb