svenhb/GRBL-Plotter #318: Grouping and combine

syrmic, 2023-01-02:
Hi Svenhb  , I wonder if you add grouping-combine tow ( plt- dxf ) or more .
[image]

svenhb, 2023-01-03:
Not easy to implement. Maybe you could change your graphics and group the objects there by selecting the same color, layer or pen width? Working for DXF and SVG, but not HPGL

syrmic, 2023-01-05:
Thanks for your reply .I have another question .how can I move g code in work my work space?

svenhb, 2023-01-06:
I don't get, what you asks for, perhaps you mean this:
You can change G-Code coordinates by using tranfrom:
[image]

If grouping is enabled, during graphics import the G-Code will be wrapped by XML-Code to seperate objects, than figures can be selected and grapped/moved with the center handle:
[image]

Wrapped code:
[image]