svenhb/GRBL-Plotter #319: Continuous, repetitive tasks,.... (I want to put a conveyor belt and vacuum table)

gorgoi, 2023-01-02:
Hello,  and thank you for this grate software!

There is any way to implement repetitive commands?
Instead of press buttons to add command in the GCODE header or somewhere else for :

**1.** Right click on mouse → **Set work coordinates on camera**
**2.** Press button->**Start automatic fiducial correction.**

[image]

**3**. Press button → **Start cutting**

[image]

**4. Repeat the cycle**

-----------------------------------------------------------
My setup, to make an idea about the work i want to do.
[image]
****

svenhb, 2023-01-03:
Do you mean hot-key function (replacing mouse-clicks by key-press)?
Or do you mean something like macro programming?

gorgoi, 2023-01-03:
Hello svenhb, and thank you!

I mean something like a function. Only to press Start button and then , repeat continuously same steps, without pressing anything.
[image]

svenhb, 2023-01-05:
Have a look into the new release: https://github.com/svenhb/GRBL-Plotter/releases
Unfortunatly you need to edit the XML file by your own text-editor - for the moment...
To set the camera work coordinate you just send G-Code "G10 L20 P0  Xxx Yyy"

gorgoi, 2023-01-06:
wow! thank you!,  I will test it this weekend and let you know

gorgoi, 2023-01-07:
Hello Seven, 

It is amazing your work, really impressive. I've made the test:

--- **example2.xml**---- Example **without fiducials worked great** for 2 hours.., and maybe more. **Super!**
----**example.xml**------**At the fiducial one nok**, the fiducials pass to rapidly, did't respect normal procedure. Maybe this will work with fixed camera, but my sheet is to big and I use XY mount one. 

There  is any way to adjust timing between in fiducials procedure?

I add it some videos and log files.

--------Fiducial NOK------------

https://user-images.githubusercontent.com/24373830/211160734-cd0ad219-4593-4252-97cd-790f0f623179.mp4

-------- Fiducial OK--------------
https://user-images.githubusercontent.com/24373830/211160636-bb9d6ad3-7bbf-47bb-a94a-df29447bd607.mp4

--------- LOG files------------------
[Logs.zip](https://github.com/svenhb/GRBL-Plotter/files/10366706/Logs.zip)

Thank you!

svenhb, 2023-01-07:
Thanks for the video, very helpful!
In second 0:39 fiducial detection starts - and finished in 0:40 - not ok.... (response "finished ok" should appear, when completly finished).
I made a new version here, which hopefully waits for the end of the detection.... Lets see...
Unfortunatly I have no hardware to check by my self...
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1_6_8_1_Setup.exe

gorgoi, 2023-01-08:
**It worked!, thank you!**
----------------------------
The only think is , if I want to stop the cycle I must press REST and Kill Alarm,  and  to start again, reopen the program, but is not an issue for me.

https://user-images.githubusercontent.com/24373830/211190235-ff3dbb4c-952a-4413-989b-d6d364c0d573.mp4

svenhb, 2023-01-08:
>The only think is , if I want to stop the cycle...
Do you mean the "Process automation" or "Fiducial detection" cycle?

gorgoi, 2023-01-08:
Thanks for response, I mean "Process automation" cycle. 

Thanks for this advice, I didn't pay attention on this, but it will help.
[image]

svenhb, 2023-01-08:
When pressing "Stop", the current step should be finished and then the automations stopps...
[image]

gorgoi, 2023-01-08:
Hm… I left the workshop, I will check tomorrow  and let you know. Thank you!

gorgoi, 2023-01-09:
You are right **it is OK**
----------------------------
I didn't had the patience to finish the cycle  and then it  stops. I was expected when I press stop to do it immediately. 
Is better how you did it, to stop at at the end of the cycle in use. SUPER!

Thank you some much, it grate so grate for repetitive tasks.

closed 2023-03-10 by svenhb