svenhb/GRBL-Plotter #320: The 1.6.8.0 version of the software runs incorrectly, and both the source and release versions have the same problem

lianzaozi, 2023-01-10:
Click on the menu bar "G-Code Creation">"Creat Text" and an error window will pop up

As shown in the picture below
[image]
[image]

svenhb, 2023-01-10:
Could you save your settings (if needed) and then reset all settings?
Hope this works...

[image]

[image]

lianzaozi, 2023-01-10:
I followed the method you said, but the same error window still appeared

svenhb, 2023-01-10:
Do you use "original" windows? Or some workarround and other OS?
The problem is: the initial font is set in the properties and properly not known by your OS:
```
    <Setting Name="createTextSystemFont" Type="System.Drawing.Font" Scope="User">
      <Value Profile="(Default)">Microsoft Sans Serif, 12pt</Value>
    </Setting>

```
You could try to close GRBL-Plotter, edit user.config manually (and set a known font)  and start again.
Location is somwhere C:\User\name\AppData\Local\GRBL-Plotter\....\1.6.8.0\user.config

svenhb, 2023-01-10:
If you compile by your own, try this (not tested) bug-fix: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter/GCodeCreation/CreateFromForm/GCodeFromText.cs

lianzaozi, 2023-01-10:
After compiling with the new code you uploaded, error Windows will still pop up during testing

[image]

svenhb, 2023-01-10:
Thanks for testing. 
Could you check the logfile for an error-entry?

svenhb, 2023-01-10:
Hm strange. It doesn't show the expected output:
When opening the form, it should log the font.
And because of error-handling, the error should not pop-up, instead also an error-log should appear...
Like:
```
2023-01-10 17:22:31.4701 | Info  | GrblPlotter.GCodeFromText      |  TextForm_Load initFont:'
2023-01-10 17:22:31.4701 | Info  | GrblPlotter.GCodeFromText      |  TextForm_Load textFont:'
2023-01-10 17:23:31.4701 | Error  | GrblPlotter.GCodeFromText      |  ShowTextSize textFont.Name:'
```

svenhb, 2023-01-15:
Try new release: https://github.com/svenhb/GRBL-Plotter/releases

svenhb, 2023-03-10:
No feedback, bug is gone?

closed 2023-03-10 by svenhb