svenhb/GRBL-Plotter #321: Commands are sent incorrectly

nfrik, 2023-01-25:
**Describe the bug**
The order of commands is incorrect. Either every 2nd one is omitted or the order is reversed.

**To Reproduce**
Steps to reproduce the behavior:
1. Generate text with G-Code creation and start plotting while looking at the real-time status report on the COM port

**Expected behavior**
1. Every command should be sent in the order of appearance in the gcode file

**Screenshots**

[image]

svenhb, 2023-01-25:
I assume you connected "Marlin" instead of a grbl-controller. Probably the first user.
The problem is: to get actual position from Marlin you have to send a "real" command M114, which also generates a "ok", this confuses my buffers...
I will have a look to fix that.

svenhb, 2023-01-25:
I can't see the problem with my Mega 2560 with Marlin 2.1.2 (just controller without real hardware), but found perhaps a point to fix.
Please try https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1_6_8_3_Setup.exe

nfrik, 2023-01-25:
This worked perfectly, thank you!

svenhb, 2023-01-26:
Good to hear, thanks for testing. 
Further feedback for usage with Marlin is welcome :-)

closed 2023-01-26 by svenhb