svenhb/GRBL-Plotter #322: Marlin stop sending commands after program end

nfrik, 2023-01-27:
**Describe the bug**
The simulator is ahead of the commands being sent that results in an incomplete pattern. For example if we want to print text "Test" in Marlin mode, only letters "Te" will be printed perhaps due to lack of sync between simulator and driver despite that coordinates are properly sent from the machine. 

**To Reproduce**
Steps to reproduce the behavior:
Create a sample single line 1-stroke text like "Test" and try to print it.

**Expected behavior**
Properly printed text.

svenhb, 2023-01-27:
Which version did you use? Did you try 1.6.8.3? https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1_6_8_3_Setup.exe

So the real movement of the machine is ok, but not the green line which shows the actual path inside the 2D view?

nfrik, 2023-01-29:
Yes, the problem was 1.6.8.3 and it looks like the there is inconsistency between the current position and position of the data sent to the machine.

Also, the GRBLplotter tries to connect to marlin every time a jog request is made.

nfrik, 2023-01-29:
Also could you please push the code for 1.6.8.3?

svenhb, 2023-01-29:
Check new release: https://github.com/svenhb/GRBL-Plotter/releases

nfrik, 2023-01-31:
Thanks, @svenhb, for the update. This version fixed most of the issues with Marlin communication, except that GRBLPlotter is still quite slow with the hardware.

The problem is due to the M114 command being sent to the machine after each movement. Even at high baud rates, the interruptions are quite noticeable.

I wonder if it is possible to set a field somewhere in the menu that would specify the number of commands that will be send to the device without position request interruptions.

svenhb, 2023-01-31:
Try this setting:
[image]

nfrik, 2023-02-02:
Thank you, that helped a bit, but there is still a problem with jogging, likely due to the buffer filling delay on the GRBLPlotter side. It takes about half a second for the G91 command to be issued. 

Also, the joystick style, with custom distance and direction might not play well with GRBL protocol since it is hard to control the position due to the need for constant feedback. Do you think it would be more practical to simplify the joystick with simple steps in X/Y/Z directions as shown in the image below?

[image]

closed 2023-05-01 by svenhb