svenhb/GRBL-Plotter #323: Simplify joystick style and remove head position requests

nfrik, 2023-02-04:
**Describe the solution you'd like**
Replace joystick with a X/Y/Z buttons with variable step sizes and eliminate head position requests. This would help to eliminate issues with communication bottlenecks and will provide more robust control of the gantry.

**Describe alternatives you've considered**
Currently, I use buttons on the driver board to move the head as the joystick is not responsive.
In some cases I set head position by writing Gcode in the serial console, but it is not convenient.

**Additional context**
Add any other context or screenshots about the feature request here.
[image]

svenhb, 2023-02-05:
You can adjust step size and speeds here:
[image]

svenhb, 2023-03-10:
Check new release: https://github.com/svenhb/GRBL-Plotter/releases