svenhb/GRBL-Plotter #324: Scan Y, extrude X feature using workpiece -> surface scan

SamuelPascal, 2023-02-16:
**Describe the bug**
While checking "Scan Y, extrude X" program sends only one line to scan, like nothing else should be checked, resulting in 1-dimensional scan

**To Reproduce**
Steps to reproduce the behavior:
1. Upload orbitrary g-code
2. Click on 'workpiece', then 'Surface scan/ Height map'
3. Set dimensions to scan to be at least 2x2 matrix
4. Check 'Prove one axis, extrude other' flag and then 'Scan Y, extrude X'.
5. See that it scans only left line from bottom to top.

**Expected behavior**
Scan the surface just like using 'Scan x, extrude Y' option (all the points that programm shows on the screen) 

**Screenshots**
1. Scan starts:
[image]
2. Scan ends:
[image]

Expected output:
[image]

svenhb, 2023-02-17:
For me the result looks correct:
You selected "Probe one axis (Y) and extrude the other (X)", which means, for one Y-coordinate all X-coordinates will get the same Z value...
[image]

SamuelPascal, 2023-02-18:
Oh, so that option only means that it scans one line and a resulting map is just "extruded" line?
I thought, there is an option to select which axis to move and probe **first** and which axis to add a step **after 1 line is scanned**. (I remebmered that in older versions second option "Scan Y, extrude X" caused to program crash..)

I really need this, as one of my axes is very poor quality and has a huge backlash.

svenhb, 2023-02-20:
I added an option to switch scan direction: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_1_6_8_5_Setup.exe

[image]

svenhb, 2023-03-10:
No response, seems to work

closed 2023-03-10 by svenhb

SamuelPascal, 2023-03-11:
Sorry, I haven't received your message in time. Thank you a lot! 
I will test that feature after I'll re-assemble my cnc).