svenhb/GRBL-Plotter #325: Edit settings and save settings in Create g code from shapes

Atlan4, 2023-03-01:
Good day. It would be possible to modify the "CREATE G code from Shapes" window so that it remembers the last used values after closing the window. Diameter, feed, x dimension... and other things that are there. I would also like a pull-out item (something like "tool:") where the last 5-10 used shapes would be stored - complete settings (diameter .... insert shape into 2D).

I miss it in MAIN setup/gcode generation, USE several pasess. Option AVOID Z-up.

Thank you

svenhb, 2023-03-03:
>so that it remembers the last used values 

Is implemented but not working right now :-(

> I would also like a pull-out item (something like "tool:") where the last 5-10 used shapes would be stored

Good idea, I will implement this feature

Atlan4, 2023-03-04:
and this "I miss it in MAIN setup/gcode generation, USE several pasess. Option AVOID Z-up." thank

svenhb, 2023-03-05:
>and this "I miss it in MAIN setup/gcode generation, USE several pasess. Option AVOID Z-up." thank

This is very dangerous if you have not a "closed" graphic, where end-point = start-point: for example an "U":
To move from end of "U" to start of "U", Z must move to safe height...

Atlan4, 2023-03-05:
I mostly make holes (simple shapes), and I go several times by 0.5 to 1 mm, and that way the Z goes up and down unnecessarily.
So it would be possible to add this option there. With some red frame and warning.

There is that option in the simple shapes tab. Well thank you

svenhb, 2023-03-10:
Check new release: https://github.com/svenhb/GRBL-Plotter/releases

Atlan4, 2023-03-13:
It seems that it works. It does not save items avoid Z, Create pocket, finally move 0,0 and insert shape into 2D.
Isn't the APPLY button useless? I have to select an item from the menu and click on Apply. It can be loaded immediately after clicking the item in the drop-down menu. Without additional confirmation APPLY?

svenhb, 2023-03-14:
>It does not save items avoid Z, Create pocket, finally move 0,0 and insert shape into 2D

This is correct, do you need them also to be saved? I thought these are more "general" settings...
I will add them

>I have to select an item from the menu and click on Apply. 

Yes i thought it would be good to think about twice before overwriting the current settings...
I will check

Atlan4, 2023-03-16:
1. Yes, it would be good to save complete settings. In order not to miss any settings.

2.Instead of apply, UNDO might be appropriate. (maybe also forward)

When you select an item, it will be uploaded immediately. But if inappropriate values ​​are uploaded, by pressing undo I return to the penultimate values ​​that were set there.

3. The diameter tool will also be added to the description of the saved values

thanks

closed 2023-04-04 by svenhb