svenhb/GRBL-Plotter #327: For enabling 4 axis support in grbl mega 5x

Girishking, 2023-03-25:
Hello sir,  xyz AB axis showing 0.000 in idle,but jogging not working , when iam jogging there is no response only showing 0.000 only in all axis.

svenhb, 2023-03-25:
What does the COM CNC window shows?
[image]

svenhb, 2023-03-25:
I tried my Mega with grbl-Mega-5X-v1.2d.20211118.hex and it works...
[image]

Whats about the joystick settings?
[image]

Girishking, 2023-03-25:
Every thing working sir, but the main problem is tangential axis not working well, $102= it's z axis, $103=820 and units per turn=3.6, it's running correct direction but, in the blade free movement between two paths in z lift position the blade rotating so many turns very fast, nema 17, 1.8° angle motor, can you pls suggest good parameters for tangential axis-A for good results....

pls fix these bugs :
1) Jogging size changes automatically too small , it's Hard operate in that moment, 
2) when iam pressing the Hard reset button AB axis is invisible, only 3 axis visible, and idle to unknown. Stops working, we have to restart it.
3) if we press the Hard reset button the every axis set to zero numbers actually (0.000). But in this version the down side zeros we can't set to zero, 
plss fix this bug sir.
In 3-axis grbl Plotter 1652 version excellent working 🤩 last i used. No bugs in that...
Any ways great job❤️,, lots of support and help from you sir, i will never forget this help, i will spend contribution From my side heartfully, i hope my tangential axis will success soon. thank you 🙏

svenhb, 2023-03-25:
If you have a real A-axis I would work with real 360° per turn. With your 1.8° motor you may set $103=200 or times 2,4,8 depending on micro-steps at your stepper motor driver.
Your list:
1) Jogging size changes automatically too small  - I don't understand the problem
2) Sound like a problem of grbl and your controller. As far as GRBL-Plotter gets the first response, it knows the amount of axis.
3) "Down side zeros" (I assume you mean the smaller numbers) are showing the machine coordinates, which will be set after homing - so you need end-switches. Or after hard-reset - it works with my controller...
[image]

svenhb, 2023-03-25:
See also https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration#100-101-102-103-104-105--xyzabc-stepsmm-stepsdegree

Girishking, 2023-03-26:
Hello sir, thank you for your reply, 
1) in my case the main problem is 'small zeros ' iam going to use endstops switches to solve the small zeros problem ,but accidentally + & - touched together in endstops pins, then Ramps gone, showing like this below 👇
[image]
its not a problem for us , we are using another ramps !
2) confusion in using endstops, i want to use normal close limit switch, bounce back to little bit opposite direction and Stops, 
This is your machine 🤩😘 I want to use endstops same like this 
[image]
I hope 'small zeros'  problem will solve after using endstops, i think !
 3) everything is fine but i want to use servo in this grbl mega 5 axis controller, where can I enable, config file or have to do any other changes?? Last time we used servomotor for pen gripper, in (Z+ for signal, 5v, ground) pins in limits switch section in cnc shiled. I want to use same pen gripper in ramps 1.4 &Arduino Mega, pls help me 🙏 for this , this is very important to me using servo motor for gripper, i connected servo to ramps but not working no response, how to enable it ??

svenhb, 2023-03-26:
1) "in my case the main problem is 'small zeros '" I still don't get it, what do you mean with small zeros?
2) you need to invert the signals: https://github.com/fra589/grbl-Mega-5X/wiki/Grbl-Configuration#5---limit-pins-invert-boolean
3) I don't know if spindle PWM can be used to control a rc-servo, better ask here: https://github.com/fra589/grbl-Mega-5X/issues

Girishking, 2023-03-26:
Now iam using 1652 version it's working excellent, bo bugs, no lags, no other issues, that down side small zeros problem solved (0.000)
[image]
1) the main problem is enabling servo motor , this is the last task for me , i have to enable the servo motor in ramps 1.4 and Arduino mega, (servo control Gcode "M3 S0" to down & "M3 S100" for up), pls help me for this , how ever i got lots of support from you,  but this is the  last task for me, 
You can only solve this problem, because your a King of grbl Plotter, plss help me for this, my humble request 🙏,  
I’ve been through the config.h spindle settings and checked cpu_map.h but nothing fixes this issue.
How ever if we are use laser module, or spindle motor, or servo motor for pen up and down, in ramps 1.4 and Arduino mega, hear the PWM signal mandatory, that is very easy to you to solve this i think,, kindly please help me for this,  plss give me some time in your your valuable time, this is very important to me 🙏pls try to understand sir,

svenhb, 2023-03-29:
You may try suggested changes: https://github.com/fra589/grbl-Mega-5X/issues/323#issuecomment-1486581884

svenhb, 2023-04-04:
Please try recommended grbl firmware changes

closed 2023-04-04 by svenhb