svenhb/GRBL-Plotter #328: Fluid-NC - coordinates not changing

kjk25, 2023-03-26:
try to use GRBL-Plotter with my Fluid-NC FW on a MKS DLC32 HW
i use a old LPKF PCB Milling Machine C30 for that.

when i move the x or y axis i don't see ToolCoordintates changing they all still 0.0 
only a short time when i klick on the jog button i see a number

GRBL Plotter Version 1.6.9.0   
VER:3.6 FluidNC v3.6.5-pre4:]
MKS DLC32 V2.1

svenhb, 2023-03-26:
Here is written it should work: http://wiki.fluidnc.com/en/features/grbl_compatibility
Whats is the response if you enable both checkboxes?
[image]

kjk25, 2023-03-26:
Hi Sven, danke for the fast response. If i can help to test please let me nkow.
I like to make my LPKF C30 workiking with manual Tool Change, i hope your programm can doit.
no other programm is able to do that feature i dont no way? 
in my cnc C30 i replaced all electronics except the spindel driver board, this is uart driven, and i sniffed the commands.
With a additional esp32 who sniff from the Sxx , M5 commands on the Gcode interface i switch the spindle on and off.  

[image]

Kai

svenhb, 2023-03-26:
The output looks correct...
How does the start-up message looks like in COM CNC window? Right after connecting to grbl?

svenhb, 2023-03-27:
I found this: https://github.com/bdring/FluidNC/wiki/FluidNC-Commands-and-Settings
You may try to set `$Start/Message=Grbl 1.1g [\H]` would give Grbl 1.1g ['$' for help]

kjk25, 2023-03-29:
when i switch it on:
> [CTRL-X] reset
RX> [MSG:INFO: FluidNC v3.6.5-pre4] 
RX> [MSG:INFO: Compiled with ESP32 SDK:v4.4.1-1-gb8050b365e] 
RX> [MSG:INFO: Local filesystem type is spiffs] 
RX> [MSG:INFO: Configuration file:kjkconfig.yaml] 
RX> [MSG:WARN: Ignored key disable_pin] 
RX> [MSG:DBG: Running after-parse tasks] 
RX> [MSG:DBG: Checking configuration] 
RX> [MSG:INFO: Machine KJK OnEh] 
RX> [MSG:INFO: Board MKS-DLC32 V2.1] 
RX> [MSG:INFO: I2SO BCK:gpio.16 WS:gpio.17 DATA:gpio.21] 
RX> [MSG:INFO: SPI SCK:gpio.14 MOSI:gpio.13 MISO:gpio.2] 
RX> [MSG:INFO: SD Card Detect gpio.39:low] 
RX> [MSG:INFO: SD Card cs_pin:gpio.15 detect:gpio.39:low] 
RX> [MSG:INFO: Stepping:I2S_stream Pulse:4us Dsbl Delay:0us Dir Delay:1us Idle Delay:200ms] 
RX> [MSG:INFO: Axis count 3] 
RX> [MSG:INFO: Shared stepper disable I2SO.0] 
RX> [MSG:INFO: Axis X (0.000,345.000)] 
RX> [MSG:INFO:   Motor0] 
RX> [MSG:INFO: Trinamic Stepper Uart Tx:gpio.27 Rx:gpio.33 RTS:NO_PIN Baud:115200] 
RX> [MSG:DBG: TMCStepper Library Ver. 0x703] 
RX> [MSG:INFO:     tmc_2209 Step:I2SO.1 Dir:I2SO.2 Disable:NO_PIN Addr:0 R:0.110] 
RX> [MSG:INFO:  X Neg Limit gpio.36] 
RX> [MSG:DBG:  X Neg Limit 0] 
RX> [MSG:INFO: Axis Y (0.000,205.000)] 
RX> [MSG:INFO:   Motor0] 
RX> [MSG:INFO:     tmc_2209 Step:I2SO.5 Dir:I2SO.6 Disable:NO_PIN Addr:1 R:0.110] 
RX> [MSG:INFO:  Y Neg Limit gpio.35] 
RX> [MSG:DBG:  Y Neg Limit 0] 
RX> [MSG:INFO: Axis Z (0.000,80.000)] 
RX> [MSG:INFO:   Motor0] 
RX> [MSG:INFO:     solenoid Pin: gpio.32 Off: 0.000 Hold: 100.000 Pull:100.000 Duration:300 pwm hz:5000 period:5000] 
RX> [MSG:INFO: X Axis driver test passed] 
RX> [MSG:DBG: X Axis StealthChop] 
RX> [MSG:INFO: Y Axis driver test passed] 
RX> [MSG:DBG: Y Axis StealthChop] 
RX> [MSG:INFO: Kinematic system: Cartesian] 
RX> [MSG:INFO: Relay Spindle Ena:NO_PIN Out:gpio.18 Dir:NO_PIN] 
RX> [MSG:INFO: Using spindle Relay] 
RX> [MSG:INFO: Probe Pin: gpio.22:low] 
RX> [MSG:INFO: Connecting to STA SSID:RadioWesterfeldSA] 
RX> [MSG:INFO: Connecting.] 
Minimize window...
Correct baud rate? Try Marlin response...
RX> [MSG:INFO: Connected - IP is 192.168.5.87] 
RX> [MSG:INFO: WiFi on] 
RX> [MSG:INFO: Start mDNS with hostname:http://fluidnc.local/] 
RX> [MSG:INFO: SSDP Started] 
RX> [MSG:INFO: HTTP started on port 80] 
RX> [MSG:INFO: Telnet started on port 23] 
RX> Grbl 3.6 [FluidNC v3.6.5-pre4 (wifi) '$' for help] 
<Idle|MPos:0.000,0.000,0.000|FS:0,0|WCO:0.000,0.000,0.000>
<Idle|MPos:0.000,0.000,0.000|FS:0,0|Ov:100,100,100>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
* Assume buffer size of 127 bytes
<Idle|MPos:0.000,0.000,0.000|FS:0,0>
<Idle|MPos:0.000,0.000,0.000|FS:0,0>

closed 2023-03-29 by kjk25