svenhb/GRBL-Plotter #329: no tool change popup dialog

kjk25, 2023-03-29:
when i run the g code no tool change dialog popup
and the spindel ignore it and execute the next command

created with CopperCAM

%
( CopperCAM - 20/03/2023 / ISO-Mill Output )
( C:\CopperCAM\test1.gcode created 24/03/2023 at 16:41 )
( Workpiece dimensions: 54.915 x 59.563 x 1 mm )
G21 G40 G54
G80 G90 G94
( Tool #1 "Universalfraeser 0.2mm" / Diameter 0.2 mm )
T1 M06
M03 S32
G00 Z1
G00 X47.496 Y16.889
G01 Z-1.2 F60
G00 Z1
G00 Y41.883
G01 Z-1.2
G00 Z1
M05
( Tool #3 "KonturCutter 1.5mm" / Diameter 1.5 mm )
T3 M06
M03 S32
G00 Z1
G00 X47.496 Y16.889
G01 Z-1.2
G01 X48.246 F300
G02 X48.246 Y16.889 I-0.75 J0
G01 X48.651
G02 X48.651 Y16.889 I-1.155 J0
G00 Z1
G00 X47.496 Y41.883
G01 Z-1.2 F60
G01 X48.246 F300
G02 X48.246 Y41.883 I-0.75 J0
G01 X48.651
G02 X48.651 Y41.883 I-1.155 J0
G00 Z1
M05
( Tool #6 "Bohrer 0.8mm" / Diameter 0.8 mm )
T6 M06
M03 S32
G00 Z1
G00 X35.914 Y4.443
G01 Z-1.2 F60
G00 Z1
G00 X25.754
G01 Z-1.2
G00 Z1
G00 Y12.063
G01 Z-1.2
G00 Z1
G00 Y22.223
G01 Z-1.2
G00 Z1
G00 Y24.763
G01 Z-1.2
G00 Z1

svenhb, 2023-03-29:
GCode, generated by GRBL-Plotter inserts a line like this for manual tool change:
`M0 (Tool:46  Color:Black (46) = 000000)`

You could try this, it will insert a pause, but no pop-up windows will show up:
Enable automatic tool change and process a script, which contains just a `M0`

[image]

[image]

kjk25, 2023-03-31:
yes that works,

script move to toolchange pos. and spindel stops.
G00 F1000 Z1 X0 Y0 S0;
M0 (Tool:1  Color:Black (46) = 000000)
i have to look into the gcode wich tool i shell put in.
after the toolchange i click play and the machine do the job.

would be cool to have this as Toolchange option with toolnumbers and popup etc. from Gcode

svenhb, 2023-04-04:
Should be fixed now: https://github.com/svenhb/GRBL-Plotter/releases

kjk25, 2023-04-07:
Fixed for the English version of the comment
[image]

kjk25, 2023-04-07:
thanks a lot for your fast response!

closed 2023-04-07 by kjk25