svenhb/GRBL-Plotter #330: Center finder rotation

Atlan4, 2023-03-31:
Good day. I need to find the center of the circular hole, the problem is that the given hole has (a slot) in the direction of the X axis. Since it is on the material of the creation, it fits well in the vice. it would be possible to add an option to the center search option: rotate the search in degrees. It is currently looking for the center by moving it in the direction of the Y-axis and the X-axis, since I have that hole in the testing place, it will find the meeting outside the real center. If it were possible to add a turning angle, it would solve my problem with complicated clamping and turning of the workpiece.

The search for the center would therefore be carried out according to the angle setting, it would be in the X axis but in an axis rotated by 10 degrees to the right (the same for the Y axis) or in the back the displacement angle for the X and Y axes would be the same (with the possibility of setting a different angle for X and another for Y - that would be a very specific thing, but maybe someone would use that option)

what do you think ? well thank you

svenhb, 2023-03-31:
Sounds good, will try to implement.

svenhb, 2023-04-04:
New release: https://github.com/svenhb/GRBL-Plotter/releases
Check, how it works: https://youtu.be/oMVjwiNikz0?t=70

closed 2023-05-01 by svenhb