svenhb/GRBL-Plotter #331: Non-functioning smoothing of the center from the outside, or where I am making a mistake

Atlan4, 2023-03-31:
How should smoothing the center from the outside work. I have a feeling that it doesn't work. Let me place it on the left, I don't understand why the possibility of searching by moving the Z axis is active (I set it to 0 before activating the center search), which is absolutely incomprehensible, the safe distance Z is ignored and after testing 1 position, Z goes to the given value (max probing distance Z and not safe distance Z, in my opinion max probing distance should be inactive). Next, position number 2 mesh is tested, but what I don't understand is the move to test position 3 (the diameter of the tested material is 150mm), so Y moves by 150mm (which is stupid since we are testing a circle), especially when the X coordinate is found to be 0 and Y is estimated as Y=0

Y should move by (diameter of tested material/2+10mm)
10mm if we consider that the center of the circle is estimated within a tolerance of +-10mm in the direction of the Y axis (we have already found the center of X), theoretically the value of 10mm could be the same as the set safe distance value. (possible solution A)

Solution B
Here I would also consider whether, instead of entering a value of 10mm, after reaching Y (diameter of the tested material/2), a window should pop up with a menu for correcting the position of the Y axis, i.e. if the operator sees that the spindle is not outside the tested edge, let him jog it to a position outside the workpiece, then confirm the continuation of the center search, the Y position is updated to the center search program and Z goes to 0, and the search for contact with the workpiece begins with Y displacement.

Solution C
(max probing distance should be inactive)
Use 2 buttons to find the outside center. center X and center Y
First for the X axis as it is now. The user jogs the probe to position 1 outside the material, the user sets the probe to Z=0 (so that it is at the height when it touches the workpiece from the side), clicks on the button to find the X center. The left edge of the workpiece is found. X goes to the safe distance, , Z goes to the safe distance, X moves by (material diameter+2*X safe distance), Z goes down to Z0, The right edge of the workpiece is found. X goes to the safe distance, Z goes to safe distance, X moves to the found center.
The user jogs the probe to position 2 outside the material. The procedure mentioned above is carried out but for the Y axis.

I like solution C. What do you think?

svenhb, 2023-03-31:
In need to check the problem.
Thanks for the suggested solutions

svenhb, 2023-04-04:
New release: https://github.com/svenhb/GRBL-Plotter/releases
Check, how it works: https://youtu.be/oMVjwiNikz0?t=83

closed 2023-05-01 by svenhb