svenhb/GRBL-Plotter #332: Error Messages in gode parsing

kjk25, 2023-03-31:
when i create with coopercam gcode it add text for the tool information like

G00 Z1
M05
( Tool https://github.com/svenhb/GRBL-Plotter/issues/6 "Bohrer 0.8mm" / Diameter 0.8 mm )
T6 M06
M03 S32

the gbrl-plotter parser reprts error in row xyz
can i prevent this anyhow?

svenhb, 2023-03-31:
The '#' sign is an indicator for a variable, unfortunatly I don't check if this sign appears inside a comment.
Will fix this.
As a workarround, don't use '#' inside a comment :-)

svenhb, 2023-04-04:
Should be fixed now: https://github.com/svenhb/GRBL-Plotter/releases

kjk25, 2023-04-07:
i use Version 1.6.9.2
in the meantime i changed coopercam language to german, then you get in the output no #
But then your parser will not recognize it, becorse i think you search for "Tool"

maybe it is possible to store the Previous line in a buffer and when the next line is a Tx Mx show dialog and comment ?
perhaps you can create a checkmark for use Toolset yes,no in settings, 
if you like to use your tools set with all parameters for the toolchange or display only the comment.

i would like to use your toolset when a Tx Mx is recognized with a dialg to change to Tool Nr..xxx

G21 G40 G54
G80 G90 G94
( Werkzeug Nr. 1 "Universalfraeser 0.2mm" / Durchmesser 0.2 mm )
T1 M06
M03 S32
G00 Z1
G00 X49.936 Y19.329
G01 F180 Z-1.2

when i change to english with the Tool it worked like expected, 
[image]

but i can't execute the toolchange script, where i move to toolchange position 
what is on my machine x0,y0,z1
would be cool to have the flexibility to use still scripts

[image]

all the best
Kai

closed 2023-04-09 by kjk25