svenhb/GRBL-Plotter #333: Clarification on M98/M99 handling

andreapicaro, 2023-04-01:
Hello, 
it is not clear to me if GRBL-Plotter can interpret M98/M99 codes and send a "flattened" commands stream to machine
or M98/M99 shall be nativelly supported by controller.
From my understanding, GRBL have no subroutines support, so I would expect that GRBL-Plotter handles them application side, yet, I tried to run a program with M98/M99 but I'm receiving errors of unsupported commands from controller.

Any clarification would be much appreciated
Thanks

svenhb, 2023-04-01:
GRBL-Plotter is interpreting M98/M99 and inserting the subroutines, before sending the gcode the grbl-controller.
There is a small bug right now: subroutine number must 2 digits - this should work:
```
(Test subroutine)
G0 X10 Y10 Z2 F2000
M98 P12       (call sub once)
G0 X30 Y10 Z2
M98 P12 L2    (call sub twice)
G0 X50 Y10 Z2
M98 P12 L3    (call sub 3 times)
M30

( Test relative moves)
O12   (start of sub)
G91
G1 Z-4
Y10
X10
Y-10
X-10
G0 Z4
X1 Y1
G90
M99     (end of sub)
```

andreapicaro, 2023-04-01:
Works perfectly, thanks

closed 2023-04-01 by andreapicaro