svenhb/GRBL-Plotter #335: Toolchange Feature

kjk25, 2023-04-09:
moved this request to other issue, sorry for the mix.

i use Version 1.6.9.2
in the meantime i changed coopercam language to german, then you get in the output no #
But then your parser will not recognize it, becorse i think you search for "Tool"

maybe it is possible to store the Previous line in a buffer and when the next line is a Tx Mx show dialog and comment ?
perhaps you can create a checkmark for use Toolset yes,no in settings, 
if you like to use your tools set with all parameters for the toolchange or display only the comment.

i would like to use your toolset when a Tx Mx is recognized with a dialg to change to Tool Nr..xxx

G21 G40 G54
G80 G90 G94
( Werkzeug Nr. 1 "Universalfraeser 0.2mm" / Durchmesser 0.2 mm )
T1 M06
M03 S32
G00 Z1
G00 X49.936 Y19.329
G01 F180 Z-1.2

when i change to english with the Tool it worked like expected, 
[image]

but i can't execute the toolchange script, where i move to toolchange position 
what is on my machine x0,y0,z1
would be cool to have the flexibility to use still scripts

[image]

all the best
Kai

_Originally posted by @kjk25 in https://github.com/svenhb/GRBL-Plotter/issues/332#issuecomment-1500050182_

svenhb, 2023-04-09:
If you use the same toolchange position for each tool, this option is for you:
It processes the given  code on pause during streaming...
[image]

kjk25, 2023-04-16:
yes this works for me, i use "G0 Z10 X0 Y0 M0 M5 M9;" for the Toolchange command

i have one issue, when the the dialog is shown and i press continue streaming the dialog disapear like expacted.
but then i have a second press for resume.
is this wanted?

please a hint how to move to a park position after project wars fonished.

best wishes
Kai

svenhb, 2023-04-16:
>please a hint how to move to a park position after project wars fonished.

Just place your code here:
[image]

svenhb, 2023-04-16:
>i have one issue, when the the dialog is shown and i press continue streaming...

You should remove the M0 (pause a running program) from your Toolchange command - the system is already in pause mode...

kjk25, 2023-04-18:
i use now G28.1 and G28 to move to toolchange otherwise 
if i move the project position 0,0 to a different location it moves to that coordinate.
thank you for all your help.
Kai

closed 2023-04-18 by kjk25

svenhb, 2023-04-19:
👍 Good idea, I was wasn't thinking about predefined positions.
https://grbl-plotter.de/index.php?id=form-coordinates