svenhb/GRBL-Plotter #338: DXF TrueType font and solid fills do not import

goscickiw, 2023-04-22:
**Describe the bug**
I'm trying to import a DXF for engraving on a plastic sign on a CNC 3018 machine with a standard conical engraving bit. The DXF has a TrueType font and a solid fill border. The font imports as a single line vector font, and only the outlines of the border are imported but it is not filled. I'm not sure if it's a bug, a missing feature or if I have some settings wrong that I haven't found yet.

**To Reproduce**
1. Open attached DXF ([test.zip](https://github.com/svenhb/GRBL-Plotter/files/11300607/test.zip)) in a DXF viewing/editing software of your choice;
2. Import attached DXF into GRBL Plotter;
3. Compare how it looks like in GRBL Plotter with how it looks like in the DXF viewer/editor. The black parts is what I expect to be cut away.

**Expected behavior**
The text should be imported as outlines of the TrueType letters, then the space inside of those outlines should be filled with tool paths to cut away remaining materials - similarly to how it works when using the "Create Text" option with hatch fill enabled. However I would prefer it to use the stepover and bit diameter values in the tool table rather than the settings in "Fill closed paths".
The border is made of two rectangles that make up the outlines, and a solid fill that should be converted to tool paths and cut away the same way as the TrueType letters.

**Screenshots**
This is how the DXF looks like:
[image]

This is how I expect the engraved plastic to look like (the plastic is white with a blue layer on top that is cut away when engraving):
[image]

This is how it imports into GRBL Plotter:
[image]

Also, when cutting away the solid fill, when the bit only has material to cut on one side, I noticed it gives best results when it moves to cut in the direction it's spinning (like on the drawing below). However when using the "Create Text" feature with the hatch fill option, I noticed it sometimes moves in the opposite direction and doesn't cut as well. I had to enable "Cross hatch" option in the settings to completely remove the remaining material, but it takes additional time. Is there an option somewhere to prevent it from cutting in the opposite direction?
[image]

svenhb, 2023-04-22:
I'm sorry, GRBL-Plotter can't do such tool paths. I would recommend estlcam.
I can't import DXF true type font, so the default is shown.
Try to export the DXF into SVG and import again, but anyway the tool path you want is "inverted" and not possible by GRBL-Plotter. 
I tried to import you DXF in LibreCad, but it looks nok.. Also import with Inkscape doesn't look ok...

svenhb, 2023-04-22:
After rebuilding your design and taking care of invertion, you would get this: but anyway not the tool path you want:
[image]

[image]

goscickiw, 2023-04-22:
> estlcam

Thanks, that seems to be closer to what I'm looking for. However it still doesn't import the text unless it's exploded first, doesn't detect the fills automatically, and requires more manual clicking than just creating the DXF before it does what I need.

This is how it looks like after generating it in estlcam and opening in grblControl.
[image]

Is there no chance that GRBL Plotter will ever be capable of generating that from the font and fill data in the DXF? After all it can generate TrueType text, hatch filled and with islands, when using the Create Text feature.
[image]

svenhb, 2023-04-22:
I will check filling DXF objects.
What kind of font do you use? If it is *.lff, you can try to copy it to the /data/fonts/lff directory.
Finally the result will look like this 
[image]
...with hatch filling, where you can adjust angle and distance - as toolpath.

But the main problem remains: you want the tool path outside of the filling regions...

goscickiw, 2023-04-22:
> What kind of font do you use?

Just the standard Arial font.

> Finally the result will look like this

That's pretty much what I'm looking for, though estlcam was able to generate the fill tool paths a bit better (it's called parallel pocketing strategy there) and seems to always cut in the right direction, but the DXF has to be imported as just outlines and all the Hole/Island, Part, Engraving settings have to be set manually for every object there.

> But the main problem remains: you want the tool path outside of the filling regions...

I'm not sure what you mean, is it the outermost rectangle that's meant to be a guide for cutting out the sign with a saw?

svenhb, 2023-04-22:
>I'm not sure what you mean

Sorry, you wrote: the plastic is white with a blue layer on top, some how I thought you need to engrave arround the letters...

goscickiw, 2023-04-22:
The plastic is similar to this: http://kbsign.com/Site/wp-content/uploads/2014/05/Plastic-Engraving.jpg
Except on this picture it looks like they used a single line vector font and an end mill, while I'm engraving Arial font with a V-bit (the end mills I have don't work well on this plastic).

This is how it looks like when engraving using the "Create Text" feature, the "Tekst" being engraved is done with cross hatch option enabled, the two bottom "Tekst" were done with just regular hatch fill so some material wasn't removed because the bit was moving in the wrong direction 50% of the time. Other engraving wasn't done by me, I'm testing on a scrap piece that had text on it previously.
[image]

svenhb, 2023-04-24:
I had a deeper look into the DXF format: I have to say I can't implement filling and truetype font.
Try to export your drawing to SVG.

closed 2023-05-01 by svenhb