svenhb/GRBL-Plotter #339: Probe one axis issue.

ScottEgan, 2023-05-08:
**Describe the bug**
When probing just one axis using the "Probe one axis, extrude other" function, Scanning along the X axis works fine, and the X dimension increments after each probe. Unfortunately scanning along the Y-axis doesn't work and the tool just stays in one position. This can be seen in the output tab gcode as well. Also in English, I think this function should say "Probe one axis, **_exclude_** other" rather than **_extrude_**.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to Workpiece > Surface Scan / Height Map
2. Click on the "Area to scan" tab
3. Select the "Probe one axis, extrude other" check box
4. Select the "Scan Y, extrude X" radio box
5. Click "Generate Height Map" at bottom of screen
6. Click the "Output" tab and review the gcode

**Expected behavior**
When scanning along X you can see in the gcode that after each probe command Z will move back up and the X position will increment. This does not happen when scanning along the Y axis.

Scan X gcode snip:
```
(Probing Size X:36 Y:26 )ExtrudeY scan X axis, extrude in Y
Code sent
G90F600
G0Z2
G0Y0
G0Z2
G0X0
G38.3Z-5
G0Z2
G0X20
G38.3Z-5
G0Z2
G0X40
G38.3Z-5
G0Z2
G0X60
G38.3Z-5
G0Z2
G0X80
G38.3Z-5
G0Z2
G0X100
```

Scan Y gcode snip:
```
(Probing Size X:36 Y:26 )ExtrudeX scan Y axis, extrude in X
Code sent
G90F600
G0Z2
G0X0
G0Z2
G0X0
G38.3Z-5
G0Z2
G0X0
G38.3Z-5
G0Z2
G0X0
G38.3Z-5
G0Z2
G0X0
G38.3Z-5
G0Z2
G0X0
```

**Screenshots**
Correct behavior on Scan X
[image]

Incorrect behavior on Scan Y
[image]
[image]

svenhb, 2023-05-08:
Thanks for the feedback. I made a mistake during code-simplification and forgot to seperate the X and Y axis.
About naming this function: I thought about CAD, generating a 3D model from a 2D scetch -> extrude. Here, 2D data is generated from a 1D scan.

ScottEgan, 2023-05-08:
Thank you for the quick response!

>About naming this function: I thought about CAD, generating a 3D model from a 2D sketch -> extrude. Here, 2D data is generated from a 1D scan.

Ahh ok. I didn't think about it that way but it makes sense now. Thank you for explaining.

svenhb, 2023-05-11:
You may try this bug fix version: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup.exe
Closing this issue happend automatically after uploading the setup.exe...

ScottEgan, 2023-05-11:
Seems to still behave the same.
[image]

svenhb, 2023-05-12:
Ok, thanks for testing. 
I thought I found the mistake as it was so obvious, without testing...

ScottEgan, 2023-05-12:
No worries, I don't mind running the tests.

svenhb, 2023-05-12:
Now it is fixed: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup.exe

ScottEgan, 2023-05-30:
Sorry for the late reply, I finally got around to testing.
Works as expected now! Thank you!
We can close

[image]

commit 7fc2be47 (2023-05-11): Version 1.6.9.5 bug fix #339
closed 2023-05-30 by ScottEgan