svenhb/GRBL-Plotter #340: Drag Tool Compensation not using start location

Rubix47, 2023-05-18:
**Describe the bug**
It does not appear that start position of the drag tool is used for the start of a cut.  When "use of drag knife" is selected, the knife is positioned to the 0 degree (to the right) at the end of a shape.  But this information is not used to compensate the blade at the begging of the next cut.

**To Reproduce**
When cutting a series of lines that are vertical the blade will be positioned to the right at the end of the cut (if "use of drag knife" is selected).  But there is no "arc" to orient the knife at the beginning of the next cut. 

**Expected behavior**
If a vertical line is to be cut and the knife is positioned 0 degrees or to the right there should be a compensation arc prior to the start of the cut.

**Screenshots**
[image]

[image]

this is an amazing piece of software and I thank you for that.  Drag knife compensation is working great everywhere but the start of cuts that cannot be started to the right (+X).  I have a need to cut very detailed stuff so this is a problem for me.  If there is already a solution I apologize, but I did not see anything in here.

Rubix47, 2023-05-18:
Thank you so much for helping and I will admit I have a very specific use for my plotter that requires probably unusual accuracy.  Most would probably not have an issue with this.  (I'm making grit blast masks for 20 sided dice)

What you showed would much improve things but the orientation of the knife would still be off for the first cut.  And this still does not solve the multiple vertical lines issue though.  each line here would start at the 0 degree (to the right) so the tip of the blade would not be on the line until some time into the cut.  
[image]
like this, pardon my art and exaugurated so you can see it.
[image]

I was trying to simply the issue with lines and maybe that's not helpful, maybe I should describe the issue as it is.  here is what I'm actually trying to cut and having issue with.  It needs to be 6mm tall(I have it a bit larger here to see things).
[image]

When it cuts the side shapes the knife is not lined up with the shapes at the start of the cut because there is not a horizontal surface to start on. in this picture the blade starts outside the shape because its at the 0 position.  what would work great is if the tip was plunged into a corner then rotated to line up with the first cut prior to cutting.

[image]

svenhb, 2023-05-18:
In principle, this option expects a knife angle of 0° on start and leaves each figure with 0° angle.
I need to check

Rubix47, 2023-05-18:
it definitly leaves each figure at 0.  But does not take full advantage of the fact that it knows where the knife tip is for the start.  Thanks for looking into it.
Cheers

svenhb, 2023-05-19:
Try this version, hope I could fix it, my tests looked good: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup.exe

Rubix47, 2023-05-19:
Yes I think that's it. looks good in my tests too.  Thank you very much!

closed 2023-07-14 by svenhb