svenhb/GRBL-Plotter #341: Source code issues with version 1.6.9.4

lianzaozi, 2023-05-24:
Click G-Code Creation=>Create Text. The following error message is displayed:"Object reference not set to an instance of an object"

[image]

[image]

svenhb, 2023-05-24:
Did you install a new font, without closing and opening GRBL-Plotter?

svenhb, 2023-05-24:
No idea. Just run the program without debugger....

lianzaozi, 2023-05-25:
> 不知道。只需在没有调试器的情况下运行程序....

Run file path "bin\Debug\GRBL-Plotter.exe" separately; Nothing unusual appears.

I can't figure out why this problem happened

svenhb, 2023-05-25:
I think it the usual behavior with debugger:
Inside the "try"-branch the error occurs, the debugger tells you this. After "Continue" the program continue in the "catch"-branch.

closed 2023-07-14 by svenhb