svenhb/GRBL-Plotter #342: create g code from shapes - wrong size hole

Atlan4, 2023-06-04:
in version 1690, I use create g code from shapes, I set the tool to 0.8mm and I want it to make an opening of 0.8mm, it is not possible to set it for x=0, it automatically chooses 0.1, which results in a larger opening than 0.8mm.
But I choose X=diameter of the cutter (x=0.8mm) and it works correctly. It would not be appropriate to automate it so that instead of the value of 0.1, which is inserted there instead of 0, it sets the diameter of the milling cutter - it would be correct with that pad, without further setting.

I assume that it is not fixed even in the newer version

It is possible to complete the export of the program settings and the shapes that are stored in create g code from shapes (which were used and were stored there for quick loading), simply to solve the overall problem when migrating to a newer version of the program so that everything is transferred there somehow easily, including the button settings for example. pen UP from MAX

well thank you

svenhb, 2023-06-04:
Yes, I can change the minimum X and Y to 0,0.
Also completing the export of the "simple shape" settings via "File - Export machine settings" I can do.
But the list of shapes for quick loading should be already kept, if a new version is installed. Check the file C;<Users\,,,\GRBL-Plotter\data\simpleshapes.xml

Atlan4, 2023-06-04:
no, if you set it to 0, it won't work. Because if you have a 0.8mm cutter and I set it to X0.1mm, the result is a hole of 0.8+0.35+0.35mm... or bigger if I choose X0.2mm, be sure to run the simulation. AZ when x0.8, ie if I enter x diameter of the cutter, it works correctly. x0.8 milling cutter d0.8, make a hole of 0.8 mm. if I give x 0.9, the milling cutter 0.8 will make a hole of 0.9 mm.

Make it so that X res Y will have the smallest value that corresponds to the diameter of the milling cutter. (ie so that x res Y cannot be entered with a value smaller than the diameter of the milling cutter)

svenhb, 2023-06-05:
You never wrote which path type you use: "On the shape", "Outside the shape" or "Inside the shape".
If I set the possible minimum for X to 0.0 and select "On the shape" it works - a slot of the width from the tool diameter will be made.
I will not check the selected path-type, to adapt the smallest X value - Too much work....

Atlan4, 2023-06-05:
Yes, it doesn't work only for inside the shapes.
[image]
[image]

svenhb, 2023-10-06:
You will get a visual warning now, if dimension is not ok:
[image]

closed 2023-10-06 by svenhb