svenhb/GRBL-Plotter #343: Error reading.plt file

lianzaozi, 2023-07-04:
Error reading.plt file

The left is read by plotter software, and the right is read by other software

[image]

[image]

Second other software read
[image]

[image]

Test software and test files

[test software and test file.zip](https://github.com/svenhb/GRBL-Plotter/files/11944063/test.software.and.test.file.zip)

svenhb, 2023-07-06:
I fixed it and also added arc and text import, so your example 'HPGL Example.PLT' also looks almost correct.
Just replace this file:
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter/GCodeCreation/ImportFromFile/GCodeFromHpgl.cs

lianzaozi, 2023-07-07:
> I fixed it and also added arc and text import, so your example 'HPGL Example.PLT' also looks almost correct. Just replace this file: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter/GCodeCreation/ImportFromFile/GCodeFromHpgl.cs

Thank you very much for your work. I have tested it and there are still some small problems
[test file and test screenshot.zip](https://github.com/svenhb/GRBL-Plotter/files/11973494/test.file.and.test.screenshot.zip)

svenhb, 2023-07-07:
Most of your new files are not matching HPGL standards  E.g. 

- line-terminator  ";" is missing, 
- number-seperator is "." not " " or ","
- unkown commands (like VS, WU, PW), which are not listed here: https://www.isoplotec.co.jp/HPGL/eHPGL.htm

I will try to import them anyway...

lianzaozi, 2023-07-07:
First of all, thanks for your reply. These.PLT files are automatically generated using software and may not fully conform to the standard plt format. Can you make an option to let the user decide whether to read non-standard.PLT files?

lianzaozi, 2023-07-07:
I have found a standard explanation book about.PLT format. I have this material, but it is not in English. I can upload it if you need. At the same time I also found this book English link https://dl.acm.org/doi/book/10.5555/523000
[image]

https://www.amazon.com/exec/obidos/ASIN/0201310147/acmorg-20

lianzaozi, 2023-07-07:
[image]

The hp website recommends developing based on this book
https://support.hp.com/ca-en/document/c01121263

lianzaozi, 2023-07-07:
> First of all, thanks for your reply. These.PLT files are automatically generated using software and may not fully conform to the standard plt format. Can you make an option to let the user decide whether to read non-standard.PLT files?

Book <The HP-GL/2 and HP RTL Reference Guide: A Handbook for Program Developers> An explanation of what you said about unknown commands
VS, speed selection
PW, brush width
WU, brush width unit selection

svenhb, 2023-07-07:
Thanks to point me to this manual: https://people.wou.edu/~soukupm/HPGL2-RTL_ReferenceGuide.pdf

>Can you make an option to let the user decide whether to read non-standard.PLT files?

I will try automatic recognition

svenhb, 2023-07-07:
It works with your sample-files: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter/GCodeCreation/ImportFromFile/GCodeFromHpgl.cs

lianzaozi, 2023-07-07:
Thank you very much for your work.
At present, it seems that the software works very well after the change, and I will conduct more tests.

lianzaozi, 2023-07-08:
What do these two buttons do?

[image]
[image]

svenhb, 2023-07-08:
During "Import vector graphic", the graphic file will be read and imported, then the importet path will be processed according selected options. Some options like "sort by distance and merge paths", hatch fill, grouping, tiling, take a lot of time.

"Process graphic objects" appears when reading and displaying the G-Code. Collecting the grouped data (XML tags), Creating the graphics-paths to display,...

lianzaozi, 2023-07-10:
I think this issue should have been fixed, and no problems were found through my other tests.

closed 2023-07-14 by svenhb