svenhb/GRBL-Plotter #345: Pen Plotter Spring Loaded Z-Axis

coopermor, 2023-07-25:
I have a Bachin T-A4 pen plotter. The Z Axis is spring loaded so turning the servo motor engages the pen with the paper, letting the motor free spool will return the pen to Z=0. I'm not sure if this is unsupported in GRBL-Plotter or if I just don't know how to configure it. I have included serial logs of sending the same SVG using both GRBL-Plotter and Bachin Draw. GRBL-Draw will properly command the XY axes but the pen does not engage the paper.

The Bachin Draw software uses 
```
$1=254
G0 Z0 F500
```
as the pen up command and 
```
$1=255
G0 Z9 F500
```
as the pen down command. I changed the custom buttons to this but I have to click twice as I get `error:20 	Unsupported or invalid g-code command found in block.` due to the GRBL command in the custom button GCode.

If you can provide any advice on how to use my spring loaded Z-axis with GRBL-Plotter I would really appreciate it.
[GRBL-Plotter.txt](https://github.com/svenhb/GRBL-Plotter/files/12166014/GRBL-Plotter.txt)
[BachinDraw.txt](https://github.com/svenhb/GRBL-Plotter/files/12166015/BachinDraw.txt)

svenhb, 2023-07-26:
How did you set the commands? It is working with my controller, without error message:
Set command either in setup, seperated by ";":
[image]

Or in seperate lines using right-click menu on button:
[image]
[image]

svenhb, 2023-07-26:
Your BachinDraw.txt shows: 
- on program start you need $1=255;$32=0; (set this in the "G-Code Header")
- on program end you need a  $1=250; (set this in the "G-Code Footer")
- as pen-up you need G1Z1F5000; set this as "Invidual command"
- also pen-down G1Z9F5000
[image]

coopermor, 2023-07-29:
Thanks this is working great.

closed 2023-07-29 by coopermor