svenhb/GRBL-Plotter #350: Xloader doesn't open

gibrie, 2023-08-21:
Downloaded GRBL-Plotter from github and unzipped. Downloaded XLoader  und unzipped. If I want to open XLoader.exe as described in GRBL-Plotter.de to select the desired .hex file XLoader doesn't open. Any suggestions?

svenhb, 2023-08-21:
Do you have all Xloader-files in one folder and started the .exe?
[image]

Which windows version do you use?

gibrie, 2023-08-22:
I work with Windows 11, all XLoader files are present in one folder (src) but what I see is that XLoader doend’t show the .exe extension.

My download unzippen shows: 
- GRBL-Plotter-1.7.0.2 containing 4 file directories and 11 various sub files
- xLoader-master-> xloader-master-> files ‘images’ +’src’ + ‘license’ + readme.md’; ‘src’ is as in your overview without the .exe-extension at the ‘XLoader’.

I will start the download over to see if the extendsion appears. Can you exactly mention the right site for the XLoader download?

svenhb, 2023-08-22:
After installation with the regular setup (https://github.com/svenhb/GRBL-Plotter/releases), you should get this folders:
[image]
I didn't upload the Xload.exe to my github, I just added to the setup.exe

gibrie, 2023-08-22:
This is the site I downloaded from and got GRBL-Plotter-1.7.0.2; xLoader has been downloade separately. GRBL-Plotter shows this: 
[image]
Firmware and GRBL-Plotter are separated, Firmware only containes the .hex-files.

svenhb, 2023-08-22:
Here is also a source: https://github.com/binaryupdates/xLoader
You need to have all files in one folder

svenhb, 2023-08-22:
And I mentioned this setup-file: https://github.com/svenhb/GRBL-Plotter/releases/download/v1.7.0.2/GRBL-Plotter_Setup.exe

gibrie, 2023-08-22:
The pop up screen of the xLoader works for now. Do you want me to enter the file in the firmware aside the hex files or do I leave them apart?
The set-up file of the GRBL plotter succeeded as well and got the short cut on my screen. So far so good!

svenhb, 2023-08-22:
You need to click on the "..." button and select the desired firmware file:
[image]
like
[image]

gibrie, 2023-08-22:
I got started with the right sites for the downloads I needed. I am on the right track and going to connect my plotter to do a try-out. Thanks a lot for your support! Excellent job!

svenhb, 2023-08-22:
Thanks for your question, so I updated my website a little bit. Hopefully the description is a little bit more clear now.

closed 2023-08-22 by svenhb