svenhb/GRBL-Plotter #352: Understanding Machine Area/Machine Limits

bryandam, 2023-08-28:
**Describe the bug**
I'm not sure if there's a better place to ask this question but I'm struggling to understand the Machine Area/Limits configuration. I've got a Bachin T-A4 pen plotter, axis are all going in the right direction, and I've already done a handful of test plots. The T-A4 has a bed size to fit US Letter or A4 paper. What I'm trying to understand is if I can configure the software to show me the working area of the plotter so I know where the text will be placed on the physical piece of paper.

I've spent a couple of nights playing with the settings in Setup > Program Behavior > Machine Area:
[image]

It definitely changes the view when 'Machine Limits' is enabled.:
[image]

However, I can't seem to get the machine area to match the actual plotter. The Bachin T-A4 has a X/Y limit switches in the top left and the screenshot above shows where the UI seems to think home is. Sometimes it shows home at the lower left of the working area. I can't figure out how to make home appear in the top left. This is important since the plotter is designed to hold paper in the top left corner and I will eventually use smaller envelopes ... so I can't make the origin the bottom right.

**Expected behavior**
I'm expecting that the upper left home location of the actual plotter match the upper left when 'Machine Limits' voew is enabled.

**Screenshots**
If applicable, add screenshots to help explain your problem.

Thanks Sven, or anyone who can she light on this. And again, sorry if there's a discussion forum somewhere I'm mising.

svenhb, 2023-08-28:
The machine area (and shown limits) is based on machine coordinates, the shown xy coordinate system shows the actual work coordinate origin.
Just move your plotter to the lower, left edge and press "Zero XYZ" to set the work origin to 0;0 at the desired position.
Your settings shown in the screenshot are looking ok.
[image]
Are both values are chaning (work and machine coordinate) when moving the machine?

[image]

bryandam, 2023-08-28:
Thanks for the response Sven.

>Are both values are changing (work and machine coordinate) when moving the machine?
Yes, they are. Should they? Only the plotter itself moves, the bed/paper is fixed in place.

>Just move your plotter to the lower, left edge and press "Zero XYZ" to set the work origin to 0;0 at the desired position.
Ok, did that and it zeroed out the Work coordinates (larger numbers) but not the Machine coordinates (smaller numbers below) but the machine area still doesn't seem to line up at all:
[image]

Note that the Home button works and sends the unit to the upper left corner where it engages the limit switches. If I do so from that lower left position after I've zero'd XYZ the physical machine moves straigth up but onscreen it's moving diagonally up and to the left. When it hits the limit switches in the top left the onscreen position 'resets' as shown here:
[image]

svenhb, 2023-08-28:
>Note that the Home button works and sends the unit to the upper left corner where it engages the limit switches. 

And at the upper left corner the _machine_ coordinates are almost 1;1?
And when you move to your desired origin (lower left corner) the _machine_ coordinates are almost 1; -290 - I assume?

So I think you should change your settings like this:
[image]

bryandam, 2023-08-28:
Thanks again for taking the time to respond Sven; that finally helped it make sense in my mind:
Machine Travel: The total distance each axis can travel.
Min/Max Positions: The relative distance from the origin (0,0) where Machine Travel is calculated from.

Part of what I was getting wrong is that the work area then is in the negative Y axis so that's the value that needed a negative min value. In any case, I have the UI and the plotter matching now. Thanks again!

closed 2023-08-28 by bryandam

svenhb, 2023-08-29:
Yes working with the machine coordinates is a bit wired, special after homing. A workarround with "manual homing" would be: you move to the lower left edge and do a hard-reset, which sets the machine coordinates to 0;0.