svenhb/GRBL-Plotter #353: The plotter software version V1.7.0.2 has a bug when reading dxf files

lianzaozi, 2023-08-30:
Version V1.7.0.2 of the plotter software encountered a bug when reading dxf files, which is not present in version 1.6.6.6 of the grbl plotter

[image]

[image]

The dxf file I used for testing
The dxf file used for testing can be read by the plotter software version 1.6.6.6
[Drawing1.zip](https://github.com/svenhb/GRBL-Plotter/files/12472375/Drawing1.zip)

svenhb, 2023-08-30:
I don't get the error, what are your settings here:
[image]

lianzaozi, 2023-08-30:
[image]

svenhb, 2023-08-30:
Can you try with 1.00? As smaller this value as more code lines will be generated. How much RAM does your computer have?

svenhb, 2023-08-30:
I don't know which other options you did enable:
With the settings above a Gcode Arc move should be generated for the circles 
(e.g. `G02 X347.773 Y510.817 I-158.000 J0.000  F800`   ).
But I assume instead the circles will be drawn by line segments. For example your "Drawing1.dxf":

1. With Arc move - ca. 100 lines
2. With "Avoid G2/3" and 1.00 "Step width" and "Remove very short moves <" 0.1 - ca. 2000 lines
3. With "Avoid G2/3" and 1.00 "Step width" and "Remove very short moves <" disabled - ca. 3300 lines
4. With "Avoid G2/3" and 0.01 "Step width" and "Remove very short moves <" disabled - ca. 327000 lines

Note: version 4 with "Remove very short moves <" 0.1 also generates 327000 objects inernally...

lianzaozi, 2023-08-31:
Setting this parameter to 1 will not result in a read failure. My computer's memory is 16GB

svenhb, 2023-08-31:
To avoid generating unnecessary data, I will limit the minimum "Step with" (internally) to the value given by "Remove very short moves" (if enabled).

svenhb, 2023-08-31:
Try this: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_b.exe

lianzaozi, 2023-09-01:
Version V1.7.0.2 of the software can only convert circles to straight lines, not G2 and G3. Whether I check "Avoid G2/3 (Arc Move) commands, draw arcs/circles with line segments" or not, the result is the same

svenhb, 2023-09-01:
On my computer it works with 1.7.0.2: "Avoid G2/3 (Arc Move)" deselect, I get G2/G3 commands, selected I get line segments.
I assume you enabled a not listed "hidden" option, which enables "Avoid G2/3 (Arc Move)" internally  (like for hatch fill).
You can try to reset all setting here:
[image]

svenhb, 2023-09-01:
With this options enabled, line segments will be created:
                ```
ConvertArcToLine = Properties.Settings.Default.importGCNoArcs || OptionClipCode || OptionDragTool || OptionHatchFill;
                ConvertArcToLine = ConvertArcToLine || OptionSpecialWireBend || OptionSpecialDevelop || OptionRampOnPenDown || OptionDashPattern;

```

lianzaozi, 2023-09-01:
Did I reset the settings according to your method or will only generate G01? Here are the settings for my computer

[image]

svenhb, 2023-09-01:
Can you do a screenshot of the status line, right after import?
[image]

lianzaozi, 2023-09-01:
I have found the issue that circles will not generate G2 and G3 because the plotter software needs to click on the "Reload all settings" button after switching the language from English to Chinese Similarly, to switch from Chinese to English, you need to click on the "Reload all settings" button; This way, the circle can generate code for G2 and G3

lianzaozi, 2023-09-01:
When using the clipping function, an unnecessary line was added

[image]

[image]

How to display the serial number of each area when using the clipping function, as shown below
[image]

svenhb, 2023-09-01:
👍👍👍 very good you found the reson. 
After switching the language, you are asked to restart the software, this should apply all settings.

>When using the clipping function, an unnecessary line was added

You showd the simulation path, which is a bit inaccurate, better try a real plot.

>How to display the serial number 

Enable the background view:
[image]

lianzaozi, 2023-09-01:
Checking 'background' does not display a serial number

[image]

svenhb, 2023-09-01:
Deüending on options it should work:
[image]

closed 2023-10-05 by svenhb