svenhb/GRBL-Plotter #354: Maintain Zoom/Position and Selection when Moving/Rotating/Scaling/Importing

bryandam, 2023-08-31:
**Describe the solution you'd like**
Currently anytime you move, rotate, scale or import something, the UI will zoom to 100% which changes the on-screen position of the object you just modified/added.  Additionally, it will deselect the object after the transform. When arranging multiple items on a larger work area this becomes really cumbersome; after every modification I have to re-zoom and relocate the thing I want to work on.

Ideally I'd like the on-screen position of the object to not move on-screen after the transform. Additionally, post transform or post addition I'd like the select object to still be selected. Selecting groups in particular can be a bit tricky so having to do it multiple times is painful.

**Describe alternatives you've considered**
Using 'Fixed Machine Area' ... kind of ... helps but _only_ if you don't need to zoom in to make really precise movements. If you zoom in to make a small movement it will reset the zoom just like above.

**Additional context**
Before:
[image]

After moving the text down just a hair:
[image]

I played around with the code tonight and I narrowed it down to:
[MainFormGetCodeTransform.TransformStart's](https://github.com/svenhb/GRBL-Plotter/blob/c9cd130ab19f3ca7c804f5529621f7fd314b2336/GRBL-Plotter/GUI/MainFormGetCodeTransform.cs#L454C16-L454C16) pBoxTransform.Reset(); call.

[MainFormGetCodeTransform.TransformEnd's ](https://github.com/svenhb/GRBL-Plotter/blob/c9cd130ab19f3ca7c804f5529621f7fd314b2336/GRBL-Plotter/GUI/MainFormGetCodeTransform.cs#L460C13-L460C40) VisuGCode.CalcDrawingArea() call. CalcDrawingArea tries to make sure that 0,0 always appears on screen which isn't good if 0,0 wasn't on screen before the transform/addition.

[MainFormPictureBox.PictureBox1_MouseUp](https://github.com/svenhb/GRBL-Plotter/blob/c9cd130ab19f3ca7c804f5529621f7fd314b2336/GRBL-Plotter/GUI/MainFormPictureBox.cs#L545) hardcodes a zoomFactor of 1.

I spent several hours tonight trying to get this to work and had _some_ success but couldn't get the zoom/position to stay the same while also successfully transforming the object.

svenhb, 2023-08-31:
I see, I will fix that. The right click menu already has an entry to set zooming to 100%...

bryandam, 2023-08-31:
>The right click menu already has an entry to set zooming to 100%
Right, but that's what I don't want to happen. If I'm zoomed in at 40% and I move/rotate something I want to stay zoomed in at 40%; not be reset to 100% zoom.

I was laying awake last night, as one does, thinking about this and I think I have an idea of why I couldn't get the code working like I wanted. Part of the problem is that the zoom factor seems to be based on the drawing size. Which can change when you rotate or resize something.  What do you think about having GCode2DViewPaths.CalcDrawingArea always use the machine limits when 'Fixed Machine Area' is enabled? That is to say, when 'Fixed Machine Area' is enabled the drawing area is also fixed and does not resize based on the objects.

I _think_ this would solve another difficulty I had in trying to make really precise placements. Because the drawing size is only that of the object on the screen you don't have the grid outside of it. So when moving something around you can't know how far you're really moving it or snap it to a grid:
[image]

Thanks again Sven, this is really great software for those of us tinkering around with plotters and the like.

svenhb, 2023-08-31:
Zoom out, to extend grid: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_b.exe

bryandam, 2023-08-31:
Again, awesome stuff, that's such a lifesaver. Transforming an object no longer resets the view.

 A couple of small things:
The grids do appear if I zoom out and back in:
[image]

However, they disappear if I transform the object:
[image]

There also seems to be an odd zoom behavior that anytime you scroll-zoom to 100% the work area is reset so that 0,0 is in the top left of the PictureWindow. This is how I came across the info window bug, because when I zoom it keeps putting stuff behind it.
[image]
[image]

Another idea that I think would be really helpful is automatically selecting the group when adding/importing an object. Similarly, when you currently transform an object it is not selected afterwards. It'd be great to have it still selected so you can more easily perform multiple transforms.

svenhb, 2023-09-01:
>There also seems to be an odd zoom behavior that anytime you scroll-zoom to 100% 

this is actually on purpose, to re-center the drawing if you accidentally leave the visible area.

Will see if I can implement your ideas.

svenhb, 2023-09-03:
Check out this version: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_c.exe

bryandam, 2023-09-05:
Beautiful work Sven. I know those changes might seem small and most people might use Inkscape for content creation but these are great quality of life improvements for people just starting out.

I noticed the groups are being reselected after transform but only if the user doesn't use one of the modifiers:
Shift when rotating (snap to degrees)
Shift when moving (snap to grid)
Ctrl when resizing (maintain ratio)

I'm not sure if there's other transform modifiers (ex. Ctrl when rotating) but if not it might be a good idea to make them all use either Shift or Ctrl.

Similarly, auto-select on creation (ex. Create Text) only works if it's the second thing to be added to the work area. That is, if you open the program and create text then the group is not auto-selected. If you create a second object (ex Text) then it's auto selected.

Lastly, one other idea that I think would be small but helpful: Add a 'New File (Ctrl+N)' option to the File menu that empties the workspace and GRBL code. It's not hard to do manually by selecting everything in the GRBL code, deleting it, and then telling it to update the work area ... but that might not be super obvious to new users.

svenhb, 2023-09-09:
Thanks again for testing and suggestions. I fixed the Shift and Ctrl problem and added a "Clear" to the right-click menu:
https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_f.exe

Usually "Add imported graphic..." is switched off and loading a new file will overwrite the current content. So I think a menu entry 'New File (Ctrl+N)' is not really usefull.
I also rearranged the right-click menu a bit
[image]

bryandam, 2023-09-12:
Thanks again Sven, finally had a moment to test. We're almost there, the zoom level and position do stay the same now when making transforms using Shift/Ctrl. However, it does not re-select the group after the transform as it does without Shift/Ctrl.

lianzaozi, 2023-09-14:
> 再次感谢你的测试和建议。我解决了移动和CTRL问题,并在右击菜单中添加了"清晰"功能: [ HTPS://吉图布.com/S文HB/GWBLBLB/GLOB/BUB/GMAT/GLOB_SETUP____________________________________](https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_g.exe)
> 
> 通常"添加导入图形。.."关闭并加载一个新文件将覆盖当前内容。因此,我认为菜单项"新文件(CTRL+N)"并不真正有用。 我还重新安排了右击菜单 [image]

> Thanks again for testing and suggestions. I fixed the Shift and Ctrl problem and added a "Clear" to the right-click menu: [...]

Thank you for adding the right-click menu "Properties of selected path" to the software. I really like this new feature and now I have some small suggestions to add the following content to the newly added properties interface:

If the selection is circle, software users can be allowed to change the radius of the selected circle in this attribute interface

2. If a straight line is selected, the user can be allowed to change the starting and ending coordinates of the selected straight line in this attribute interface, which means adding the option to change the starting and ending coordinates of the straight line in this interface

lianzaozi, 2023-09-14:
> Thanks again for testing and suggestions. I fixed the Shift and Ctrl problem and added a "Clear" to the right-click menu: [...]

Thank you for adding the right-click menu "Properties of selected path" to the software. I really like this new feature and now I have some small suggestions to add the following content to the newly added properties interface:

1) If the selection is circle, it can allow software users to change the radius of the selected circle in this attribute interface, which means adding the option to change the radius of the circle in this attribute interface

2) If a straight line is selected, the user can be allowed to change the starting and ending coordinates of the selected line in this attribute interface, which means adding the option to change the starting and ending coordinates of the straight line in this attribute interface

svenhb, 2023-09-16:
The selected path can be figure, group, tile, collection... To complicate to change a single parameter from there.
>If the selection is circle, 

Just drag the upper left arrow and press "Ctrl" to keep ratio
[image]

>If a straight line is

Just do a "Alt" click to select the end of the line (you get a small rectangle at the node) and move it
[image]

closed 2023-10-05 by svenhb