svenhb/GRBL-Plotter #355: Info Box Is Not Removed When Disabled

bryandam, 2023-08-31:
**Describe the bug**
The Info Box is sometimes poorly placed in the top right of the work area. Maybe there's a way to relocate it but I couldn't find how to do so.
[image]

If you disable this in the View options it just becomes plain white box. As does the work position box by the cursor:
[image]

**To Reproduce**
Using 1.7.0.2
Steps to reproduce the behavior:
Disable View > Info Box

**Expected behavior**
Info Box is entirely removed from the work area UI when that option is disabled.

svenhb, 2023-08-31:
I never checked this combination (Show machine limits, hide info). 
Will fix this.

svenhb, 2023-08-31:
Is fixed here; https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_b.exe

bryandam, 2023-08-31:
Awesome, can confirm this is fixed. 
Thanks Sven!

closed 2023-08-31 by bryandam