svenhb/GRBL-Plotter #356: Problems to import spindle values from tool table in case of PWM servo

Pancholina, 2023-09-04:
I have built a plotter with a spring between the servo arm and the pencil. In this way, I can use different servo positions to change the force applied to the pencil, and obtain different gray levels.
I'm trying to import a SVG or a DXF file with different colors that should be converted to different gray levels by the different spindle values in the tool table.
I was able to do it in fast test that I did few weeks ago (during the assembly tests) but now that all the assembly job is finished I cannot reproduce.
I don't know if this is because now I'm using different version of the program (now I'm using the last one) or because I'm doing something wrong now.
I have read too (https://github.com/svenhb/GRBL-Plotter/issues/191) that  spindle values from tool table are "only for real spindle commands, not RC-Servo pen-up/-down".
The questions are:
Is possible to use the spindle values from the tool table for PWM servo?
Which would be the way to do?

Thank you in anticipation.

Rafa

svenhb, 2023-09-04:
Yes it should work, as long as you are using the correct grbl-firmware (https://grbl-plotter.de/index.php?id=quick-guide&setlang=en#pwm).
Depending on selected tool the according S-value will be sent.

BTW. I made a new option to convert the pen-width from the SVG or DXF to S-value: https://grbl-plotter.de/index.php?id=form-setup-1#path-interpretations
Unfortunatly I have no example setup....

svenhb, 2023-09-04:
To get your tool table working, you need to 

- enable the use of the tool table
- get spindle speed from tool table
- Adapt the tool table itself

[image]

[image]

[image]

Pancholina, 2023-09-04:
Thank you for your fast response.
Yes. It works!!!
My mistake was to select  "PWM" instead of "Spindle/ laser" option at "Pen up / down translation". In this way only speed was read, not spindle.
Thank you again.

svenhb, 2023-09-04:
How does your servo react on "M05" spindle off?
In my case the servo moves to one end-position and I had change the servo mounting, to get the pen lifted on that end position.

svenhb, 2023-09-04:
I think I haven't thought about this case: getting S-value for PWM from tool table.
The best would be to have "from tool table" also on the PWM-tab for pen-down to keep the delays.

Pancholina, 2023-09-04:
This would be great.
Do you think that this could be added in the next revision?

Thank.

Rafa

svenhb, 2023-09-04:
Yes, I just started to check the code, where to implement.

Pancholina, 2023-09-04:
Wonderful !!!
When are you planning to release the next revision?

Thanks a lot.

svenhb, 2023-09-04:
This evening  - test version 1.7.0.2.d. 
Or if you can't wait and want to build your own release in visual studio: Up to now I just changed GCodeCreation\Graphic2GCodeRelated.cs function "GetValuesFromToolTable"  line 1625 to 
`GcodeSpindleSpeed = GcodePwmDown = toolInfo.SpindleSpeed;`
And I need to copy the Checkbox "Spindle speed - from tool table" to the PWM tab
Then this should work 
[image]

svenhb, 2023-09-04:
Please try https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_d.exe

Pancholina, 2023-09-05:
I have tested with SVG and DXF files and works perfect. I think this is a good improvement for the program and for sure very useful for my plotter.
Thank you.

svenhb, 2023-09-05:
👍 thanks for testing

closed 2023-09-09 by svenhb