svenhb/GRBL-Plotter #357: `M300` G-Code stops streaming

merny93, 2023-09-04:
**Describe the bug**

Some plotters, such as the the XPlotter, use a `M300 SXXX` G-Code to control a servo motor z-axis (admittedly a interesting design decision). GRBL-Plotter interprets the `M300 SXXX` G-Code as a `M30` plus some extra bits and immediately stops streaming.

**To Reproduce**

Stream any G-Code file which includes a `M300 S90` code. The G-Code syntax highlighting even shows the `M30` bit in yellow.

**Expected behavior**

`M300` should not be parsed as `M30` and streaming should proceed.

svenhb, 2023-09-05:
I would say it would be an enhancement...
Will try to implement

svenhb, 2023-09-05:
Try this version: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_e.exe

closed 2023-10-05 by svenhb

merny93, 2023-12-23:
Thanks!

Sorry I did not get a chance to test it. My Windows machine gave up on me and I switched to a linux based tool :(. 

Great work on the software tho! Helped me a lot to get set up on my new machine!