svenhb/GRBL-Plotter #359: Reading ellipses stored in blocks can result in incorrect results

lianzaozi, 2023-09-09:
The software version 1.7.0.2 reads elliptical arcs stored in blocks, which may result in incorrect orientation of the read graphics. This error should exist in all versions

test file
[test.zip](https://github.com/svenhb/GRBL-Plotter/files/12564347/test.zip)

test image 
On the left is the result read by version 1.7.0.2 software, and on the right is the result read by CAD software
[image]

svenhb, 2023-09-09:
If I draw such ellipse in LibreCAD, the import in GRBL-Plotter is correct.

lianzaozi, 2023-09-09:
I used autocad 2019 to draw this figure, and the elliptic arc is in a block. If the elliptic arc is not in the block, it is also correct for plotter to read the elliptic arc that is not in the block

svenhb, 2023-09-09:
Found the problem: I applied group-angle also to start/end angle of ellipse.
Try https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_f.exe

lianzaozi, 2023-09-11:
There is also an error reading arcs stored in the block

Plotter software and AutoCAD software read the arcs stored in the block, and the coordinates of the starting point of the arcs displayed by the two software are different

[image]

test file

[dxf.zip](https://github.com/svenhb/GRBL-Plotter/files/12575145/dxf.zip)

svenhb, 2023-09-11:
Should be fixed now: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_h.exe

lianzaozi, 2023-09-12:
After testing, version 1.7.0.2-g of the software temporarily found no issues when reading elliptical arcs and arcs stored in blocks

svenhb, 2023-09-12:
👍 I hope you mean 1.7.0.2-h

lianzaozi, 2023-09-13:
Sorry, there was an error in the input of the previous reply version

Yes, the version I tested is 1.7.0.2-h, and no issues have been found so far

closed 2023-10-05 by svenhb