svenhb/GRBL-Plotter #360: Drag and drop SVG with 'Add imported graphic to 2D-view' enabled makes the load process loop.

Anacapala, 2023-09-10:
**Describe the bug**
Drag and drop SVG with 'Add imported graphic to 2D-view' enabled makes the load process loop.
Windows 10, GRBL-Plotter Version 1.7.0.0.

**To Reproduce**
1. Start GRBL-Plotter and enable the option at the bottom of the main screen to 'Add imported graphic to 2D-view'.
2. Close GRBL-Plotter.
3. Drag and drop a .SVG file onto the GRBL-Plotter icon or menu item.
4. The main screen loads, the start-up overlay loads, then the 'Select Use Case' panel loads repeatedly.  The multiple copies can be closed, but in some cases new copies continue to load. Closing each instance of the panel causes the Import Information pane to appear briefly. Mutiple identical copies of the GCode are created.

**Expected behavior**
The 'Select Use Case ' panel should load and the loading process should stop and wait for user input. Only one GCode listing should be created.
Note 1: As this happens with Drag and Drop, and multiple drop files are not supported, there would never be an existing graphic for the new graphic to be added to.   
Note 2: I believe it would be acceptable for the 'Add imported graphic to 2D-view' option to be disabled at each shutdown. It would therefore need to be explicitly re-enabled, if required, after the first image had been loaded.
Note 3: If Drag and Drop is upgraded to support multiple drops then the behaviour for the case of multiple drops with the 'Add' option disabled would need to be defined.  Suitable behaviour would be to treat multiple drops as if the 'Add' option had been enabled.

**Screenshots**
If applicable, add screenshots to help explain your problem.

svenhb, 2023-09-10:
Strange issue, but is fixed now: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_g.exe
I also added multiple file import, if  'Add imported graphic to 2D-view' is enabled. If this option is disabled, just the fist file will be imported.
Objects will be arranged first in x axis until max number reached, using the values from this option - good idea?:
[image]

Anacapala, 2023-09-10:
"but is fixed now:"
Confirmed.  Drag and drop no longer causes a looping on load.

"I also added multiple file import, if 'Add imported graphic to 2D-view' is enabled."
Confirmed. Works as described.  If the option is enabled and multiple files are selected for drag and drop onto an icon or menu item then the files are displayed and loaded exactly as they would be if they had been selected one by one after GRBL-Plotter has been started. 

If the Multiply option in 'Setup\Path import\Path addons' is enabled those multiplications are applied to each of the imported images. 

Note that the multiply option multiplies along both the X and Y axes, if those values are set and the option is selected.  This is not quite what you describe, but is correct IMHO.

"If this option is disabled, just the fist file will be imported."
Confirmed.  Works as described.  Note that 'first' in this context is OS and operator dependent. For instance, in Windows the last file selected comes first.  It has always been like this: the app cannot tell how multiple files were selected.

For my use case the 'Add imported graphic...' option will add date-printed information to each drawing.  Each day I create a new date SVG, then I import it with any image printed that day.  It saves having to edit each image with a new date.   I imagine a similar usage would apply to logos, borders, headings, watermarks and anything similar which needs to be in the more than one image but might get updated from time to time.  An arbitrary limit on the number of files selected at one time might be worthwhile to protect the user from gross selection errors.

svenhb, 2023-09-11:
It was the first try, Perhaps I will add some options how to handle added files.
One important thing is an extra XML-Tag to separate the new file-content from the existing.

svenhb, 2023-09-11:
> Each day I create a new date SVG, then I import it with any image printed that day.

You could also add the date via "Create Text" using a Windows or Hershey font...
[image]

svenhb, 2023-09-11:
Some improovment with multiple import: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_h.exe
Also multiple import on drag&drop on already running program.
New XML-Tag "Collection" to separate code from different files.
Object selection toggles between figure, group, collection, none - with each mouse left-click

closed 2023-10-05 by svenhb