svenhb/GRBL-Plotter #361: DIY Pen Plotter with Automatic Tool Changer | CNC Drawing Machine

yifandianzi, 2023-09-14:
[image]

May I ask how to set the tool table above the Y-axis? After setting, the pen position in the tool table is negative, and all the tool tables in the pictures are in the negative direction of the Y-axis

svenhb, 2023-09-14:
Tool table positions must be set in machine coordinates - homig of the plotter is needed (or manual "homig" via hard reset on lower left position):
[image]

[image]

svenhb, 2023-09-14:
Just change the 277 to 0

yifandianzi, 2023-09-14:
[image]

https://github.com/svenhb/GRBL-Plotter/assets/94118847/46c40b4b-b3ee-48c3-ac87-54c015b7c436

Hello, this is a demonstration set to 0

svenhb, 2023-09-14:
You are right, there is a bug and the offset is not shown, but processed during streaming. Will fix it soon.

svenhb, 2023-09-14:
You may try this version: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_i.exe

yifandianzi, 2023-09-15:
https://github.com/svenhb/GRBL-Plotter/assets/94118847/1228b2b6-13a1-491b-bb9c-f4277c031af0

Hello, there is another issue now, which is that it can return to its normal position when returning to zero. When starting to work, the Z motor will rotate in the opposite direction, and then it can surprise the next step normally

svenhb, 2023-09-15:
I don't exactly understand this issue. I assume the problem is the servo-position, when PWM is off.
Compare behavior of PWM on/off here:
[image]

Compare behavior of PWM low/high here:
[image]

svenhb, 2023-09-15:
Now I could watch your video on a PC - it's not a servo...
I thing you need to "Zero Z" your Z axis before starting streaming.

yifandianzi, 2023-09-16:
[image]
Hello, I have reset all X, Y, and Z here to zero, but when starting work, they still rotate in the opposite direction. I think it may be due to incorrect settings elsewhere, and it should not be rotated without selecting a pen

svenhb, 2023-09-16:
It is normal that the Z axis move to a save position on start.
Example code after importing a simple line:
[image]

Line 14 G00 Z2.00 move Z upwards to save position
Line 17 move to start position
Line 18 G01 Z-1 move Z down, to start drawing
Line 19 draw the line
Line 20 G00 Z2.00 mive Z upwards

yifandianzi, 2023-09-16:
[image]
[image]
Did I make the right changes here

svenhb, 2023-09-16:
I don't know your hardware, I didn't saw a servo for a gripper...
Do your gripper opens with M3 S100?

Did you check my site? https://grbl-plotter.de/index.php?id=auto-pen-change&setlang=en

svenhb, 2023-09-16:
Looks like the Z-stepper should turn for a half turn maximum to move from lowest Z to highest Z.
Did you setup $102 correctly?

yifandianzi, 2023-09-16:
Should I set $102 correctly to $40?
https://www.youtube.com/watch?v=ZxeStwRespc&list=RDCMUCJHZY4Tl6ovsnBrNocdDbAg&start_radio=1&rv=ZxeStwRespc&t=65
https://www.youtube.com/watch?v=The following link is mine_4S9PE

svenhb, 2023-09-16:
Last video is not visible. I don't know your driver settings for micro-steps.
Usually a stepper-motor needs 200 steps for a full turn, If I interpret your design correctly, you have a half turn for about 20mm Z-change. With full-step-mode you may set $102 = 5 (100 steps/half-turn divided by 20 mm)
If you have half-step mode: $102 = 10

If your Z axis is spring loaded to keep it up, you may have a problem with step loss at the Z motor.

yifandianzi, 2023-09-17:
https://www.youtube.com/watch?v=ZJZuB_4S9PE
Hello, setting $102=10 and $102=5 will prompt an error, while setting $102=40 will not cause an error

yifandianzi, 2023-09-17:
I have set these three places in the firmware
[image]

Do you think it's correct

svenhb, 2023-09-17:
I don't understand how your z-axis works, in your video time 0.04, the stepper is moving, but the z-wagon is not moving....
Also I don't see an end-switch for z. So you can't use homing for z.
You need to set: #define HOMING_CYCLE 0 ((1<<X_AXIS) | (1<<Y_AXIS)) if you have end-switches there

yifandianzi, 2023-09-17:
https://www.youtube.com/watch?v=NDZsdDLbDDo
Normally, it should start working after resetting to zero, and then X Y starts moving to the pen selection position and Z starts lifting. The problem now is that when the start button is clicked, the Z stepper motor has already started working, and then the Z limit is set off

yifandianzi, 2023-09-17:
https://www.youtube.com/watch?v=ZxeStwRespc&list=RDCMUCJHZY4Tl6ovsnBrNocdDbAg&start_radio=1&rv=ZxeStwRespc&t=65
The author's point is that the Z-lift motor does not work when starting work after resetting to zero

svenhb, 2023-09-17:
The main problem is, that pen selection positions are machine coordinate, which needs homing.
Also the linked video shows a working z-homing - you need to change your mechanics...
[image]

yifandianzi, 2023-09-17:
Thank you for your great help. When you pointed out the location, I also raised the Z limit position to zero. Now, just as I was preparing to start the work, the Z stepper motor continued to raise and then raised again, triggering a limit fault

yifandianzi, 2023-09-17:
https://www.youtube.com/watch?v=QQNdMw_KsNo
After all X Y Z are reset, the execution starts at 0:53, and the Z stepper motor continues to rise

svenhb, 2023-09-17:
After Z-homing, Z stays at top position which you may want to have at +2mm in work coordinates.
After Z homing move 2mm down and press "Zero Z".
Also important the choosen travel way for import - small enough for your hardware?:
[image]

yifandianzi, 2023-09-18:
[image]
Hello, where is the white area set and where can I reach the coordinates

yifandianzi, 2023-09-18:
https://www.youtube.com/watch?v=qdFgPm4bOkc
This is my complete operation steps, but there are still problems and I don't have a clue now

svenhb, 2023-09-18:
Looks good so far.
1. set your travel ranges via $130, 131, 132. https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#130-131-132--xyz-max-travel-mm
This will influence your homing machine coordinates:
[image]

After this you go to setup and take over the ranges:
[image]

svenhb, 2023-09-18:
Here you need a value, close to your homing Y-value = ca. 350:
[image]

yifandianzi, 2023-09-19:
[image]

Hello, where else do I need to set the Z stepper motor to continue rising after all XYZ resets? I set it to zero in the picture

svenhb, 2023-09-19:
The problem is here:
[image]
Y-Offset is 191, Tool-2 Y-pos is 172, so the machine tries to move to Y=(191+172)=363 - out of your limit.

You can check this box, to see sent commands:
[image]

svenhb, 2023-09-19:
Perhaps you are not using the latest bug fix, with the tool-offset:
[image]

https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_i.exe

yifandianzi, 2023-09-20:
Hello, this is my demonstration process. The X-axis has stopped before reaching the position of the pen holder. Do you need to set the X position in the picture
https://www.youtube.com/watch?v=qE-if9f868A
[image]

svenhb, 2023-09-20:
>The X-axis has stopped before reaching the position of the pen holder

Your homing x machine coordinate is 348:
[image]

For pen change, the the X-pos will be used:
[image]

So the plotter is moving to machine-X =0 at minute 0.45:
[image]

Note: from here it moves 18mm into the pen-holder because of the script:
[image]

Either you add an offset here: 
[image]

Or you adapt the grbl travel range $130 https://github.com/gnea/grbl/wiki/Grbl-v1.1-Configuration#130-131-132--xyz-max-travel-mm

To avoid drawing on pen-holder:
Move your plotter manually (with joystick) to this point and press "Zero XY"
[image]

yifandianzi, 2023-09-20:
Thank you for your help. Now we can move on. Another question is: the graph to be drawn and the text to be written start from XY zero point. How can we move to the center of the plotter to draw? There are also tool tables 1, 2, 3, 4, 5.... Where should these positions be adjusted?
[image]

svenhb, 2023-09-20:
>How can we move to the center of the plotter to draw? 

Set up your 0;0 on the green cross (as shown above)
Load a graphic with Offset removal
[image]

Then set your center coordinates here (e.g. 180;90) and "Apply Offset":
[image]

>There are also tool tables 1, 2, 3, 4, 5.... Where should these positions be adjusted?

Just measure your pen holders...
[image]

yifandianzi, 2023-09-21:
Hello, the problem I am currently facing is with the software, which cannot be displayed at all. I tried to install the software by changing the computer system, but after opening it, it still remains the same. It cannot be maximized, and some settings also cannot see the screen    Using the latest version
https://www.youtube.com/watch?v=15d6JGEen88
[image]

svenhb, 2023-09-21:
Should look like this (https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_i.exe):
[image]

[image]

svenhb, 2023-09-21:
[image]

yifandianzi, 2023-09-23:
Hello, do you still need to set up the process of picking up and lifting the pen
https://www.youtube.com/watch?v=uYTOFpzlZCA

svenhb, 2023-09-23:
Looks good so far.

- You may adjust the servo positions for open and close - see https://www.youtube.com/watch?v=uFfjXCRGSnM
- You may disable PWM-off when finished, to avoid strange servo position:
[image]

yifandianzi, 2023-09-23:
https://www.youtube.com/watch?v=WWQc27jp7ks
Thank you. After picking up the pen, the servo released it and it got stuck while writing. Where can I cancel this step

svenhb, 2023-09-23:
I don't know where the last movement comes from.
- from footer setup?
[image]

Switch on this check box and watch sent commands:
[image]

yifandianzi, 2023-09-23:
Hello, I have tried to modify these places, but they still don't work and I will loosen the pen
[image]

yifandianzi, 2023-09-23:
In this way
[image]

yifandianzi, 2023-09-23:
https://www.youtube.com/shorts/gHtDPaVl4kA
It's Like That

yifandianzi, 2023-09-23:
I am using a special version of firmware with servo support
https://github.com/cprezzi/grbl-servo

svenhb, 2023-09-23:
I made a mistake in my last post...You need to use Z axis and disable spindle use.
[image]

yifandianzi, 2023-09-23:
Hello, is that the case
[image]

yifandianzi, 2023-09-23:
Hello, I just tried to turn on the Z-axis and disable the spindle like that, but the effect is still the same. I will release the pen

svenhb, 2023-09-23:
Can you show the generated gcode

yifandianzi, 2023-09-23:
Hello, is that the case This is preparing to write after zeroing
https://www.youtube.com/shorts/GNgGmXgAvh8

svenhb, 2023-09-23:
I mean a screenshot like this
[image]

yifandianzi, 2023-09-23:
[image]
In this way

svenhb, 2023-09-23:
And the last lines of the code?

svenhb, 2023-09-23:
If you watch my video https://youtu.be/tVS8A3_E2IY?t=141
the last action is, removing the pen, then end of program "M30" which causes the servo to move...

yifandianzi, 2023-09-23:
[image]
Hello, it's from the 27-47 paragraph

yifandianzi, 2023-09-23:
Execute M30 only after complete writing
[image]

svenhb, 2023-09-23:
Try new version https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1_7_0_2_k.exe
It removes the final M05 in line 1354

yifandianzi, 2023-09-23:
[image]
Hello, my version you sent me will still have the same issue after installation

svenhb, 2023-09-23:
The M05 should be gone:
[image]

There is still M30 which is needed. You can try to edit the code mnually to see what happens:
Set edit mode:
[image]

remove last line, then update 2D-view:
[image]

[52 more replies]