svenhb/GRBL-Plotter #364: Application launch issue

carre82, 2023-10-29:
The application start is "interrupted" by Windows 10.

It has been working properly for a while and doesn't start anymore today.

Steps to reproduce the behavior:
1. Click on the launch icon or .exe file
2. Look on the task manager
3. GRBL Plotter appears in the list, then its status switch to "interrupted" and disappear.

svenhb, 2023-10-29:
Which version? How did you install the program? Via setup? Installation as "Admin" or for current user?
Did you try to reinstall via setup? Is a controller already connected?

carre82, 2023-10-30:
Hello Sven
I installed version 1.7.0.3.c   .exe file from github, as admin, for all user, Windows 10. I reinstall after registry cleaning. doesn't work with and without controller connected.

Then I tried to install on another PC and ... it works ! 

I fear that some windows update creates the trouble ... looking at the event journal, I have a log of issue with .NET framework and a second just after with Kernelbase.dll.

I hope you can help even if the issue seems to come from my configuration.

svenhb, 2023-10-30:
I use WIn 10 Home, Version 22H2 with latest updates, without problems. I usually install it as current user, not admin. Can you try this?

carre82, 2023-10-30:
Yes, I will. But the previous installation worked well for a time and then stopped yesterday.

carre82, 2023-10-30:
removed, cleaned, reinstall as current user : it works ! Many thanks for helping ! your app is the best I tried and I would have been really desappointed to use another

closed 2023-12-18 by svenhb