svenhb/GRBL-Plotter #366: The new version (V1.7.1.0) of the software has introduced some new bugs

lianzaozi, 2023-12-18:
1. The software may encounter errors when reading polyline primitives stored in blocks that have been rotated and scaled during insertion into dxf

2. When the software scrolls the mouse wheel to zoom in and out of the displayed graphics, or double-click the mouse wheel to make the graphics adaptive, the process is very sluggish

3. After opening the "1. dxf" file and then opening "2. dxf", the graphics of "1. dxf" and "2. dxf" will be displayed on the interface at the same time. The normal working process is that the software should clear "1. dxf", then load and display the content of "2. dxf"

4. The software displays an error while reading the file "2. dxf"

[test file and screenshot.zip](https://github.com/svenhb/GRBL-Plotter/files/13705263/test.file.and.screenshot.zip)

svenhb, 2023-12-18:
About point 3, please read about multiple file loading: https://grbl-plotter.de/index.php?id=form-setup-4#file-loading

lianzaozi, 2023-12-20:
> About point 3, please read about multiple file loading: https://grbl-plotter.de/index.php?id=form-setup-4#file-loading

Thank you very much for your reminder

svenhb, 2023-12-21:
Try new version: https://github.com/svenhb/GRBL-Plotter/releases/download/v1.7.1.0/GRBL-Plotter_Setup_1710a.exe

lianzaozi, 2023-12-21:
> Try new version: https://github.com/svenhb/GRBL-Plotter/releases/download/v1.7.1.0/GRBL-Plotter_Setup_1710a.exe

Thank you.

Version 1.7.1.0a of the software did not find any issues when parsing scaled and rotated entities stored in blocks, but now there are new issues: ①. Double clicking the mouse wheel shape does not adapt ②. The process is very slow when scrolling the mouse wheel to zoom; Version 1.7.0.2 of the plotter software does not have issues ① and ②; I guess question ① and question ② should have been introduced after version 1.7.0.2

lianzaozi, 2023-12-22:
I have discovered some new issues where the shortest path sorting effect is not as good as version V1.6.6.6

Screenshot of shortest path sorting for version v1.6.6.6
[image]

Screenshot of shortest path sorting for version v1.7.1.0a
[image]

version 1.7.1.0a software Circles and arcs can only generate lines, but not G02 and G03
[image]
Parameter settings for version 1.7.1.0a software
[image]

svenhb, 2023-12-22:
About sorting:
In 1.6.6.6 the start point was XY = 0;0
Now the start point is XY = 0; maxY - some how it was better for my needs

About missing circle:
You use unusually large coordinate values, together with a bug, the circle was not drawn...

Try new version: https://github.com/svenhb/GRBL-Plotter/releases/download/v1.7.1.0/GRBL-Plotter_Setup_1710b.exe

lianzaozi, 2023-12-22:
> About sorting: In 1.6.6.6 the start point was XY = 0;0 Now the start point is XY = 0; maxY - some how it was better for my needs
>
> About missing circle: You use unusually large coordinate values, [...]

Version V1.7.1.0b of the software can only generate G01 for circles and arcs, and cannot generate G02 and G03
![111](https://github.com/svenhb/GRBL-Plotter/assets/53146195/e04fc739-45be-4bdf-a037-9160de
[image]
7e1aa6)

lianzaozi, 2023-12-22:
> ①. Double clicking the mouse wheel graphic cannot adaptively display ②. The process is very slow when scrolling the mouse wheel to zoom; 

Version 1.7.1.0b still has issues ① and ②

svenhb, 2023-12-22:
>Version V1.7.1.0b of the software can only generate G01 for circles and arcs, and cannot generate G02 and G03

Perhaps you have some options enables, which prevents using G02/G03?
Check lower left edge for information: 
[image]

svenhb, 2023-12-22:
>Double clicking the mouse wheel graphic cannot adaptively display

I didn't implement a function for double clicking the mouse wheel... What should happen?

>The process is very slow when scrolling the mouse wheel to zoom;

Perhaps because the the coordinates such large? Try to scale down the original DXF to perhaps 1000 mm (instead of 200000 like now). 2nd reason could be: when zooming out the ruler grid will be extended.

lianzaozi, 2023-12-22:
"11. gif" is a screen recording generated using version 1.7.6.6 software,

"22. gif" is a screen recording generated using version 1.7.1.0a software,

Observing the "11. gif" display screen recording, it can be observed that when I use the mouse scroll wheel to zoom in or out of the trajectory (the trajectory display exceeds the display range when zoomed in, and becomes very small when zoomed in), I quickly and continuously press the mouse scroll wheel twice, and the trajectory display will automatically return to the center of the canvas, and adaptively display it on the canvas. This feature has existed in previous versions, but has been lost in recent versions after upgrades

By comparing the screen recordings displayed in "11. gif" and "22. gif", it can be found that regardless of whether the operation is zoomed in or out, version 1.7.6.6 of the software does not lag during this operation process;

Watching the "22. gif" display screen recording, you can find that scrolling the mouse wheel to zoom in and out of the displayed trajectory, and quickly double clicking the mouse wheel to adaptively display the trajectory in the middle of the canvas, this operation process is very slow

You can personally operate it and compare the differences between the two versions of software, and you will find the problem I described

"11. gif" is a screen recording generated using version 1.7.6.6 software,↓↓↓↓
[image]

"22. gif" is a screen recording generated using version 1.7.1.0a software↓↓↓↓
[image]

svenhb, 2023-12-22:
About not using G02/G03 Code - you enabled "Dashpattern" this prevents using G02/G03
[image]

svenhb, 2023-12-22:
About zooming: in older version the view (offset) was resetted, when reaching zoom 100%. In current version not. Now you can use right click menu "Reset zooming".

lianzaozi, 2023-12-23:
Can resetting zoom be associated with double clicking the mouse wheel? Double clicking the mouse scroll wheel to reset zoom is a common practice in many Cad software operations

During the process of scrolling the scroll wheel to zoom in and out of the displayed trajectory in version 1.7.1.0a of the software, the zooming in and out speed becomes very slow. You can refer to the zooming in and out speed in "11. gif" for reference

lianzaozi, 2023-12-23:
After unchecking "Process dated line path", circles and arcs can generate G02 and G03, but a line that is not in the dxf file appears (the line pointed by the red arrow in the image)
[image]

svenhb, 2023-12-23:
>Double clicking the mouse scroll wheel to reset zoom is a common practice in many Cad software operations

Should work now

https://github.com/svenhb/GRBL-Plotter/releases/download/v1.7.1.0/GRBL-Plotter_Setup_1710c.exe

lianzaozi, 2023-12-25:
Thank you for your hard work

The software version 1.7.1.0c has the function of quickly double clicking the middle mouse button to initialize the display of trajectories, and the problem of reading straight lines that are not in dxf for multiple segments has been solved

Currently unresolved issues: software that displays trajectories when scrolling the middle mouse button to zoom in and out, or when pressing the left mouse button to move the displayed trajectory, may experience lag; The software takes a lot of time to read and store dxf files for many entities. Version 1.6.6.6 of the software performs well in reading dxf files for many entities, and the zooming and moving display trajectory does not lag

lianzaozi, 2023-12-29:
When reading a large dxf file, version 1.7.1.0a of the software froze, resulting in a blue screen crash on the computer; Using the same dxf file and version 1.6.6.6 software, the reading speed is very fast and efficient.

Test dxf file
[111.zip](https://github.com/svenhb/GRBL-Plotter/files/13790323/111.zip)

svenhb, 2023-12-29:
Are you sure the settings are the same, when using version 1.6.6.6 or version 1.7.1.0?
During import, you may see what the delay causes:
[image]

You may switch off this option, to speed up the import:
[image]

Also disabling display options can help to speed up import and zooming behavior:
[image]

lianzaozi, 2024-01-02:
The settings of the two software are the same

closed 2024-03-16 by svenhb