svenhb/GRBL-Plotter #367: RESET - waiting for response of grbl controller

Helle62, 2023-12-31:
Hello,
before, I wish you a happy new year!

I use the Arduino DUE with GrblHAL and if I want to work with GRBL-Plotter, I get the message "RESET - waiting for response of grbl controller" and the status is "Unknown".

The serial monitor shows:
* Open COM12
> [CTRL-X] reset
Correct baud rate? Try Marlin response...

I can see and configer GRBL with the app "ioSender". ioSender give me no error message.

I search the bug two days but I have no more idea what I can test to run GRBL-Plotter.

I hope someone can give me a tip.

Best regards,
Helmut

svenhb, 2023-12-31:
Happy new year, for you too :-)
Did you try to perform a hard-reset?
- either pressing the "RST" button from the COM-window
[image]

- or using the hardware reset button on the controller

Some how GRBL-Plotter is missing the welcome message from grbl.: https://github.com/gnea/grbl/wiki/Grbl-v1.1-Interface#grbl-push-messages

But I have to say, I never tried GrblHal, perhaps it does'nt send a welcome message?

svenhb, 2023-12-31:
Here I found some information about the missing welcome message in GrblHal: https://github.com/grblHAL/core/wiki/For-sender-developers#connecting

Helle62, 2023-12-31:
Many thanks for the fast response on this day!
If I use RST or the hardware reset button, I get the mesage andthe controller does't come back: 
[image]

Here are the welcome mesage from ioSender:
[VER:1.1f.20231229:]
[OPT:VNSLW2,100,1024,4,0]
[AXS:4:XYZA]
[NEWOPT:ENUMS,RT+,HOME,SED]
[SIGNALS:RHS]
[FIRMWARE:grblHAL]
[NVS STORAGE:*FLASH]
[DRIVER:SAM3X8E]
[DRIVER VERSION:231228]
[BOARD:Ramps 1.6]
[AUX IO:3,3,1,0]
[G54:0.000,0.000,0.000,0.000]
[G55:0.000,0.000,0.000,0.000]
[G56:0.000,0.000,0.000,0.000]
[G57:0.000,0.000,0.000,0.000]
[G58:0.000,0.000,0.000,0.000]
[G59:0.000,0.000,0.000,0.000]
[G59.1:0.000,0.000,0.000,0.000]
[G59.2:0.000,0.000,0.000,0.000]
[G59.3:0.000,0.000,0.000,0.000]
[G28:0.000,0.000,0.000,0.000]
[G30:0.000,0.000,0.000,0.000]
[G92:0.000,0.000,0.000,0.000]
[HOME:0.000,0.000,0.000,0.000:0]
[TLO:0.000,0.000,0.000,0.000]
[PRB:0.000,0.000,0.000,0.000:0]

I have a secont test system with GrblHal - BlackPill and CNC-Shield. This works well with GRBL-Plotter!
Welcome message at GRBL-Plotter:
* Open COM4
> [CTRL-X] reset
GrblHAL 1.1f ['$' or '$HELP' for help]
* Assume buffer size of 255 bytes

Many thanks again! Now we will wait on the better year 2024 ;-)

Have a wonderful evening!

svenhb, 2023-12-31:
Ok, some how your 1st controller doesn't send the welcome message.
GRBL-Plotter is looking for the keyword "'$' for help" to switch to grbl mode.
Perhaps you can upload the hex-file for the duo, which I can use with the XLoader to program my bare Duo-Board?
So I can try to fix it, next year....

Helle62, 2024-01-01:
Happy new year!

Sorry, I have only a BIN-file to flash my Arduino DUE. I created this file with the grblHAL Web Builder -> [http://svn.io-engineering.com:8080/](url). And to flash my Arduino I use the app "Bossa" - https://www.shumatech.com/web/products/bossa
 
I'ts not allowed to upload a BIN-file here, so I have renamed it as TXT-file.
[arduinoDUE_Ramp16_USB.txt](https://github.com/svenhb/GRBL-Plotter/files/13805326/arduinoDUE_Ramp16_USB.txt)

VG. Helmut

svenhb, 2024-01-01:
Thanks I found also the BIN generator, but was hoping to get a "ready to upload" hex...
I will try at the weekend and report.

svenhb, 2024-01-06:
Too bad: "Bossa" doesn't regonize my Duo-Board. COM Port is listed in Bossa but connection not possible (Could not connect to device on COM3).
I read a lot about this problem but didn't find a solution...

Compiling and uploading grbl-Mega-5X via Arduino IDE works...

Helle62, 2024-01-06:
Thursday night, until 3 o'clock in the morning, I tried to create a HEX file with the Arduino IDE, but I didn't manage it.
The next problem is, even if I create a HEX file, I still don't have any configuration for it.

This are the three steps from the Bossa webside. This I use to flash my both Arduino DUOs, without troubles:

First off I recommend erasing the Arduino Due using these steps:
1) Power the Due through the USB-micro cable attached to the **“NATIVE USB”** port
Hold down the erase button
2) Press and release the reset button while continuing to hold down the erase button
Release the erase button
3) Check that the Arduino now appears in device manager as “Bossa Program Port (COM…)”

Thank you for your invested time, you can close now this thread.

Happy weekend,
Helmut

svenhb, 2024-01-06:
I read this also, but...
- What or where is the USB native port?
- Where I find the erase button? My Arduino Mega board just has an reset button...
- BTW : I have a Mega 2560 isn't it the same as a Duo?
- My board always appears as "USB-Serial CH340"

svenhb, 2024-01-06:
Too stupid of me... Of course the DUE is something completely different than the Mega2560... No wonder it doesn't work with the Bossa tool...

Helle62, 2024-01-06:
Ohh sorry, I think the Mega 2560 is not listet in grblhal. Grblhal is only for 32 Bit prozessors

Helle62, 2024-01-06:
No problem, I will search a other solution. If I found a other way, I let you know.
Many thanks again!

svenhb, 2024-01-06:
Don't know why, I though I have a DUE. But on the board is written Mega 2560...
So I can't test grblHAL for the moment...

closed 2024-01-06 by svenhb