svenhb/GRBL-Plotter #368: Tool path questions

Stoli123, 2024-01-01:
I don't have a machine hooked up yet. Just checking out the software to see if it will be  good fit for my pen plotter project. So far I have been looking at the image import feature and running simulations. Import a black and white bitmap. Desired output is a stippling style print. Plot direction set to horizontal. The resulting g-code always starts the plot in the upper left, moves back and forth across every line until it gets to the bottom (y=0) . G-code looks correct. 

1. Is there a way to change the direction of the path? For instance, can I start at the bottom and work up?
2. The pen always moves across the entire line even when there is no activity (pen down). How can I eliminate the g-code where there is no pen activity? 

Thank you. It's really impressive so far.

svenhb, 2024-01-01:
Hi,
here you fin a bit more info (I need to update...): https://grbl-plotter.de/index.php?id=form-image&setlang=en
1) this option are avaiable:
[image]
>For instance, can I start at the bottom and work up?

You may try to use "from 2view" with your own pattern.

2) just disable "Only left to right"

> Desired output is a stippling style print. 

Do you know the other converters?
https://grbl-plotter.de/plotterfun/
https://grbl-plotter.de/plotterfun-width/
https://grbl-plotter.de/plotterfun-color_2020/

Stoli123, 2024-01-01:
Thank you. I think I need to read more about the pattern feature so I can better understand how they are used. I think the answer to both of my questions is there. I tried "from 2D view" with some of the example patterns but the result was not desirable. 

Thank you for the reply. Could you provide a link to information on patterns?

I understand the "Only left to right" feature, but that does not prevent the tool from traveling to the far left and right extents even if there is no pen up/down movement.  Here is a pic that hopefully explains what I am talking about. Even though the pen is not used until Y=500, the tool path includes complete horizontal passes on Y=517 down to to Y=501 (shown in green). 

[image]

svenhb, 2024-01-01:
You are right, the tool always moves to the far left or right. I will check if can add an option to prevent this behavior.

>Could you provide a link to information on patterns?

Inside the subfolder ...GRBL-Plotter\data\examples you find some pattern_*.* files, which you can select from here
[image]

Perhaps this video helps: https://www.youtube.com/watch?v=7Nvf4UwK_yk&t=522s

Stoli123, 2024-01-01:
I'll check out he video when I get a chance. Thanks again!!! 

I'm getting really excited about this project and look forward to using GRBL-Plotter. From what I have seen so far, I think it will be a good fit.

svenhb, 2024-01-01:
Sounds good.
I will try to help and modify code if needed and useful.

Stoli123, 2024-01-20:
If you choose to incorporate such a feature, that would be great. However, it would be fairly easy for me to create a script in python, or whatever, that removes the extraneous movement between the pen ups and pen downs. I don't plan on using GRBL Plotter to send the g-code directly to the machine. I will probably just use it to create the g-code files which I will upload to the machine with a different program. Having to run a script on the files before they are uploaded would be fairly easy and quick.

svenhb, 2024-01-21:
Try this test version: https://github.com/svenhb/GRBL-Plotter/blob/master/GRBL-Plotter_Setup_1710d.exe
I use an other algorythm now for line by line and added more options:
[image]

svenhb, 2024-02-03:
Thank you for the praise. I also translated some algorithms from Python and Javascript, which was sometimes not so easy...

Stoli123, 2024-02-06:
Quick question... It's probably staring me in the face, but how do I reverse the direction of a plot? If I open an SVG file with one long continuous path, it seems to start at the nearest point to the origin. Is there an option to plot the path in reverse?

closed 2024-03-16 by svenhb